sidoh/esp8266_milight_hub #610: Color carryover following change of bulb_mode

waparo, 2020-03-31:
### Describe the bug

### Expected behavior

A bright white flash is not expected when I send a command to set light to a dull salmon colour.
#### Firmware version

### I am happy to try to set up a simpler reproducible package.  
The salmon color is used for sensor controlled night lighting and the initial flash of white light is a bit disconcerting at 2am!  I have searched for other similar experiences but have not found a solution. Thanks, Chris, for any help.

sidoh, 2020-03-31:
Hi,

Can you paste your settings?

waparo, 2020-04-01:
Chris, I seem to have made a complete mess of my post and would appreciate your deleting it.
I will start again with a new post.

sidoh, 2020-04-01:
Oh, the HTML comments I had in the template are eating what you wrote.  I see it now, don't worry about creating a new post.

sidoh, 2020-04-01:
I think probably what's going on here is that Milight bulbs maintain separate brightness levels for color and white modes.  So setting the brightness when the bulb is in white mode does not affect the brightness when you switch to color mode.

If you want to avoid being blinded when the light is turn back on, set the bulb to whatever mode you'll end up in.  You might be trying to do that with the `bulb_mode` attribute in the second command, but that attribute is read-only (this is confusing -- sorry).

I'd recommend changing your second command to something like this:

```json
{"status": "Off", "color":"20,7,0","brightness": 0}
```

That way the bulb will be very dim when it's first turn on in the 3rd command.

waparo, 2020-04-01:
Chris, 
That has certainly stopped the annoying white flash.  But now I am seeing an annoying salmon flash as the white flash turns off!!  I can work on that using your suggestions.  Thank for your help.  I have been struggling with this problem for a long time.

sidoh, 2020-04-01:
You're most welcome!

Milight doesn't make stuff like this easy, and some of it is impossible.

Gonna close since it sounds like this answered your question, but do let me know if I've missed something :)

closed 2020-04-01 by sidoh