sidoh/esp8266_milight_hub #612: Can white get added to payload for MQTT state topic pattern?

dink911, 2020-04-08:
Can white get added to payload for MQTT state topic pattern when using hex_color?

When I change color everything is updated correctly because color is available but if I use white no color available (ie.. color:#FFFFFF)

Currently:
Color payload
{"state":"ON","status":"ON","brightness":255,"saturation":100,"color":"#0400FF"}
White payload
{"state":"ON","status":"ON","brightness":255}

Thank you for your time and efforts this product is amazing.
Steven G

sidoh, 2020-04-11:
Hi @dink911,

You can add the `computed_color` field under "Group state fields" in MQTT settings.  This will use the nested object format (e.g., `{"color":{"r":255,"g":"255","b":255}}` instead of `{"color":"#FFFFFF"}`), but it will get included when the bulb mode is set to white.

Not presently possible to include the computed color in hex format.  Not impossible to add, but it'd be a little unwieldy.  

Does `computed_color` give you what you need?

dink911, 2020-04-12:
I have played with the different color options and figured out a way to make it work, but program is looking for hex format so would have made it easier if white showed up like below as well.

{"state":"ON","status":"ON","brightness":255,"saturation":100,"color":"#FFFFFF"}

No worries thanks for the reply
Steven G

sidoh, 2020-04-14:
Makes sense.  Glad you got it working.

closed 2020-04-14 by sidoh