sidoh/esp8266_milight_hub #618: Hub doesn't reconnect to WiFi automatically

TroLoos, 2020-04-18:
### Describe the bug

My MiLight Hub doesn't reconnect to WiFi by itself. Sometimes I have to reconfigure my WiFi network (UniFi system), and after short unavailability of WiFi, my Milight Hub gets lost.

### Steps to reproduce

Turn off WiFi, turn on WiFi after short while.

### Expected behavior

MiLight Hub reconnects to WiFi after network is available again.

### Setup information

#### Firmware version

1.10.5 (nodemcuv2)

#### Output of http://milight-hub.local/about

```json
{"firmware":"milight-hub","version":"1.10.5","ip_address":"10.0.20.100","reset_reason":"External System","variant":"nodemcuv2","free_heap":16800,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```json
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"IPADDRESS","mqtt_username":"xxxxxxxxxx","mqtt_password":"xxxxxxxxx","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[26548,28790],"gateway_configs":[],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{"Bright_Bathroom":["rgb_cct",26548,4],"Bright_Dinner_Table":["rgb_cct",26548,2],"Bright_Lounge":["rgb_cct",26548,1],"Bright_Master_Bedroom":["rgb_cct",26548,3],"Color_Bathroom":["rgb_cct",28790,4],"Color_Dinner_Table":["rgb_cct",28790,2],"Color_Lounge":["rgb_cct",28790,1],"Color_Master_Bedroom":["rgb_cct",28790,3]}}
```

### Additional context

sidoh, 2020-04-18:
Mine all reconnect after a wifi disconnect.

Anything interesting in serial logs?
Does the setup AP appear?

TroLoos, 2020-04-18:
> Mine all reconnect after a wifi disconnect.
> 
> Anything interesting in serial logs?

What do you mean by serial logs? I can share it with you, let me know how to do it.

> Does the setup AP appear?

No, it doesn't reset itself, just disappears. To reconnect I have to power cycle NodeMCU. Then it all goes back to normal.

sidoh, 2020-04-19:
You can do it through the Arduino IDE.  I prefer a terminal with `platformio device monitor`.

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html

Gotcha.

Could be.  All of the WiFi reconnection stuff is handled by the SDK (i.e., below EspMH).  It works on both my router (dd-wrt) and my phone's hotspot.  Probably doesn't hurt to have a more aggressive reconnect, but would want to see that there's some precedent.

Millennium-bug-82, 2020-05-06:
I have the same issue with my D-Link dap-2695 AP. I've tries all kinds of stuff to optimize the AP but nothing helps.

roadsnail, 2020-06-16:
Also see the same behaviour on my esp8266 milight hub. Ubiquiti  AP-AC-Lite in use here too.

stefanheijnen, 2020-06-17:
This is caused by the SDK version used. I'd have the same issue with all my esp8266 clients. Also using Unifi products. Every brand works better with a different SDK version. 

See this issue for more info: https://github.com/arendst/Tasmota/issues/5469

sidoh, 2020-06-17:
@stefanheijnen, if I'm reading the issue correctly, there's only a difference when Tasmota's "[dynamic sleep](https://tasmota.github.io/docs/Dynamic-Sleep/)" functionality is enabled. The stats in the issue show 0 disconnects for all tested versions when dynamic sleep isn't enabled.

I'm pretty sure this project is using several of the features from the current version of the SDK it's on, so it's not a quick thing to revert to an older version. You are, of course, welcome to give that a try. Given how old 2.4 is (its initial release predates KRACK), not sure it's the best idea, though.

I _think_ releases older than 1.6 are built against 2.3, but would have to check.

Xento, 2020-09-06:
I have the same problem and i'm using Unify APs as well.
Maybe there could be a workaround for example reset the esp when there is no connection for a minute or something like that?

stefanheijnen, 2020-09-08:
I must say that I no longer noticed issues with my hubs. Firmware 1.10.6. My Unify AP's are on 5.x.

sidoh, 2020-11-24:
Gonna close this since it doesn't seem to be a consistent issue for many people -- please feel free to respond if it's still a problem and we can re-open.

closed 2020-11-24 by sidoh