sidoh/esp8266_milight_hub #621: Wifi stability problems 1.10.6

poudenes, 2020-04-24:
Hi All,

After I updated from 1.10.5 to 1.10.6 I have lots of wifi stability problems. 
Changed from N to G didn't work. I reverted back to 1.10.5 and now it work again on G Wifi Mode.

I cant send logs etc anymore, because my hub is active in the house... Don't have a second module to test on as well..

sidoh, 2020-04-26:
hi @poudenes,

Check out the 1.10.6 diff, there's nothing obvious that would cause this behavior:

https://github.com/sidoh/esp8266_milight_hub/pull/617/files

I've been running 1.10.6 myself since the release and have had no connectivity problems, so I can't reproduce.

Will need your help in the form of more details to investigate further.

poudenes, 2020-04-26:
I will try to update it again. Only what I can send are some json outputs...
My partner gets crazy when the house is not working again... I spend to much time with automation, etc... :( otherwise I can send you all the info...

sidoh, 2020-04-26:
Settings may help, really just need to be able to reproduce the issue.

What are the exact symptoms?  Are you sure the ESP8266 drops off of your wifi network?  Does it respond to pings, is the web UI available, etc?

poudenes, 2020-04-27:
Wemos drops from Wifi, try to access webUI. at one moment it load, when do refresh its gone... I need to reset the Wemos then it boot and connect again.

Sometimes take long time before it react. With 0.10.5 this is all gone...

sidoh, 2020-04-27:
I see.

Output of `/settings` and `/about` would be helpful (make sure to scrub passwords).

poudenes, 2020-04-28:
I've change some things on my router as well: instead of combine all wifi modes 2.4ghz, 5ghz in 1 ssid. I have now 2 ssid's for each wifi mode. So connect Wemos to the 2.4ghz. 

Let see if this will help as well. I will come back later on

poudenes, 2020-05-01:
no issues anymore at all... not sure what it was, but the issues are gone.. ;)

closed 2020-05-01 by poudenes

poudenes, 2020-05-01:
I had a power outage and since then all problems are back

/settings

```
{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.100.24:1883","mqtt_username”:”XXXXXXX”,”mqtt_password":"XXXXXXX","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:hex_device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:hex_device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":100,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"packet_repeat_throttle_sensitivity":50,"packet_repeat_throttle_threshold":500,"packet_repeat_minimum":50,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"Fast blip","led_mode_operating":"Slow toggle","led_mode_packet":"Fast blip","led_mode_packet_count":4,"hostname":"MilightHub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"192.68.100.X”,”wifi_static_ip_gateway":"192.168.100.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":50,"home_assistant_discovery_prefix":"","wifi_mode":"g","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[43681,43682,43683,43684,43690,43691,43692,43693,43694,43695,43706,43707],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type"],"group_id_aliases":{}}
```

/about

```
{"firmware":"milight-hub","version":"1.10.6","ip_address":"192.168.100.70","reset_reason":"Software/System restart","variant":"nodemcuv2","free_heap":14472,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

Tontze, 2020-05-01:
I Wonder does it have anything to do with this found from smartthings forum:

Turns out that the data pin grounds are isolated from the power ground… this makes good sense in many perspectives as most folks will want to tie thier grounds down their own way, but for radio… its death.

YOU MUST have a ground plane for this to work if your device is powered from a grounded power source.

SO how? simple tie your VIN ground directly to your radio ground pin… ALSO to your Ground data pin… ALL THREE must be connected.

This makes the radio dead on stable… zero packet drop, really strong gain… and if you have a good USB charger, the range is pretty far…

poudenes, 2020-05-02:
@Tontze Thanks for your input. I use a Wemos D1R2 board. Powered by USB. Use a Anker USB Power hub because use several RPi as well. The power cord is not grounded in wall switch.

Can you assist me how to connect it for a Wemos D1R2, I will give it a try.

I saw already the VIN pin :) but then I'm lost what to do