sidoh/esp8266_milight_hub #622: OFF command with remote updates hue,saturation,mode_bulb... and get out of sync

Faexa, 2020-04-28:
### Describe the bug
Hello!
First, thank you for your wonderfull work with this retro-engineering ! Very helpfull getting a state feedback with my milights.
My problem is the following:

When I change mode or color with the milight_hub everything is working. With MQTT I can store the values sent 
ex:
{"state":"ON","brightness":255,"level":100,"hue":103,"saturation":62,"bulb_mode":"color"}

But now, if I send OFF command with a remote (the one with 8 channels, hue, saturation and brightness): the state gets updated with the last known parameters within the remote (which the bulbs does not get since they are shuting down.
After OFF command, here is the state:
{"state":"OFF","brightness":255,"level":100,"kelvin":0,"color_temp":153,"bulb_mode":"white","effect":"white_mode"}
(note the bulb mode changed in this example: because with te remote I've set white, and with the hub I changed later to color)

If I turn ON the lights again with the remote, the lights are colored (since the remote didn't send hite command to the bulbs), but my MQTT show this :
{"state":"ON","brightness":255,"level":100,"kelvin":0,"color_temp":153,"bulb_mode":"white","effect":"white_mode"}

And therefore the status is out of sync with the real status of the lights.

To sum it up:
Receiving an OFF command from a remote should not update the status with what is sent by the remote since it won't update the lights.

Thank you !
Best regards

sidoh, 2020-05-01:
Hi @Faexa, thanks for the detailed report.  Can you paste a packet log captured while you're reproducing the issue?  (This is the output you see in the Web UI after clicking "Start Sniffing")

sidoh, 2020-11-24:
Going to close since this has been open for so long. Feel free to respond w/ the packet capture and I can re-open.

closed 2020-11-24 by sidoh