sidoh/esp8266_milight_hub #624: Issuing commands to multiple lights from Home assistant with a single command

strumf666, 2020-05-01:
### Describe the bug

With increasing number of lights it becomes impossible to issue a command to all at the same time.

### Steps to reproduce

Problem started exhibiting when I added more than 7 lights (I think, since I added more since then). Controlling one at a time isn't a problem, but running a script to set all of them (or a large enough number of them) on, off change brightness, effect does not work. Apparently only some of the lights successfully receive the command and execute it. Repeating the command a few times usually works, but this isn't exactly desirable. 

### Expected behavior
Not having to repeat the command, obviously :)
I am guessing a delay between sending the commands would fix this behavior, which could be implemented on either end, but considering some commands take longer than other I think it would be better to let the milight hub handle this.

### Setup information

#### Firmware version
1.10.6 but this isn't limited to just this version

#### Output of http://milight-hub.local/about

```json
{"firmware":"milight-hub","version":"1.10.6","ip_address":"192.168.1.31","reset_reason":"Software/System restart","variant":"d1_mini","free_heap":14760,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":413}}
```

#### Output of http://milight-hub.local/settings

```json
{"ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":70,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.1.50:1883","mqtt_username":"mqtt","mqtt_password":"56667","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":250,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Off","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"n","default_transition_period":500,"rf24_channels":["MID","HIGH"],"device_ids":[1,2,3,4,5,11,22,33,34,55,65,81,97,99],"gateway_configs":[],"group_state_fields":["state","brightness","mode","bulb_mode","computed_color","effect"],"group_id_aliases":{"Dnevna_soba":["cct",65,3],"Dnevna_soba-RGBW":["rgbw",3,3],"Kino":["cct",55,3],"Kino-RGBW":["rgbw",1,1],"Kopalnica":["cct",22,2],"Kuhinja":["cct",33,1],"Kuhinja-RGBW":["rgbw",2,2],"Kuhinja2":["cct",34,1],"Mamina_soba":["cct",81,3],"Mamina_soba-RGBW":["rgbw",5,4],"Postelja":["cct",99,4],"Pred_soba":["cct",11,1],"Spalnica":["cct",97,3],"Spalnica-RGBW":["rgbw",4,4]}}
```

### Additional context

sidoh, 2020-05-06:
Can you please include more detail about your setup? How many groups do you have? How are you issuing the commands?

Looks like your hub is dropping packets:

```
{"queue_stats":{"length":0,"dropped_packets":413}}
```

which means it's getting too many packets to buffer in memory at some point. The only solution here is to increase the size of the buffer (which you can do by compiling the firmware yourself by setting `MILIGHT_MAX_QUEUED_PACKETS` in [platformio.ini](https://github.com/sidoh/esp8266_milight_hub/blob/master/platformio.ini#L39) to something higher than 20), or to send fewer packets at a time.

strumf666, 2020-05-07:
```
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":70,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.1.50:1883","mqtt_username":"mqtt","mqtt_password":"56667","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":250,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Off","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"n","default_transition_period":500,"rf24_channels":["MID","HIGH"],"device_ids":[1,2,3,4,5,11,22,33,34,55,65,81,97,99],"gateway_configs":[],"group_state_fields":["state","brightness","mode","bulb_mode","computed_color","effect"],"group_id_aliases":{"Dnevna_soba":["cct",65,3],"Dnevna_soba-RGBW":["rgbw",3,3],"Kino":["cct",55,3],"Kino-RGBW":["rgbw",1,1],"Kopalnica":["cct",22,2],"Kuhinja":["cct",33,1],"Kuhinja-RGBW":["rgbw",2,2],"Kuhinja2":["cct",34,1],"Mamina_soba":["cct",81,3],"Mamina_soba-RGBW":["rgbw",5,4],"Postelja":["cct",99,4],"Pred_soba":["cct",11,1],"Spalnica":["cct",97,3],"Spalnica-RGBW":["rgbw",4,4]}}
```
I am issuing command from home assistant, if I am doing it for each light there is no problems, if I want to turn on/off, set brightness,.. for all the lights or a lot of them, only some respond to the first command, and some more to the second,... If I repeat the command enough times all lights execute the command.

Is it possible to include this under settings in the next build? I am not too comfortable nor experienced with compiling.

sidoh, 2020-05-07:
Sure, can consider it. There's not a ton of free heap to play with, though, so going way higher is asking for trouble.

btw - you could experiment with reducing the number of packet repeats. This means each packet will be sent faster, so you'll effectively have more throughput. It might help with dropped packets.

strumf666, 2020-05-10:
I rewrote my scripts and added a delay between each light and the issue is gone, as expected, but the minimum delay I can set is 1 second which makes the execution of scripts much longer than I would like. 
I'll test how much lights I can send command to at once, before packet loss occurs.
Regarding packets, how many packets are needed for a command? Always 1:1 or does it depend on a command? I am guessing that on/off require only one packet per light, but what about effect, brightness, color change?

sidoh, 2020-11-24:
Woof, sorry I missed this.

The bulbs only need to receive packets once. The repeats are a crude way of improving responsiveness.

Going to close because there are other issues open about making packet queue length configurable.

closed 2020-11-24 by sidoh