sidoh/esp8266_milight_hub #625: Switching cct mode to rgb mode - unwanted flash

waparo, 2020-05-03:
This is a sequel to my query #610/611.  Following that query I have resolved my problems with brightness fluctuations.  They were due to overlapping transitions.  No surprises there once I thought about it.  My continuing problem is that I cannot successfully set both rgb brightness and cct brightness to zero in preparation for issuing a new command.
Here is my test sequence:
{"status": "On", "brightness": 180, "color_temp": 500}
light turns on with warm white.
7 seconds later the following commands are sent
{"status": "Off", "brightness": 0, "color_temp": 154}
{"status": "Off", "brightness": 0, "color": "20,7,0"}
The warm white light is replaced by a flash of cold white followed by no light.
7 seconds later the following command is sent.
{"status": "On", "brightness": 2, "color": "20,7,0"}
There is a brief flash of white followed by the desired dull salmon color.  The flash of white is unwanted and unexpected.

You suggested that "probably what's going on here is that Milight bulbs maintain separate brightness levels for color and white modes. So setting the brightness when the bulb is in white mode does not affect the brightness when you switch to color mode".

That seems to be what is happening in my example.  I have tried to separately reset the white and color modes in the second and third commands but clearly the white mode has not been reset. 
So my question is how can I successfully set brightness of both modes to zero?

Firmware versions 1.10.5 and 1.10.6

Thanks, Chris, for any help or clues you can give.

sidoh, 2020-05-06:
The second command in this sequence won't have an effect:

```
{"status": "Off", "brightness": 0, "color_temp": 154}
{"status": "Off", "brightness": 0, "color": "20,7,0"}
```

Bulbs do not respond to anything other than the "on" command when they're off.

You should get the desired effect if you remove `{"status":"Off"}` from the first command.

waparo, 2020-05-07:
Chris, I followed your suggestion:
{"status": "On", "brightness": 180, "color_temp": 500}
{"brightness": 0, "color_temp": 154}
{"status": "Off", "brightness": 0, "color": "20,7,0"}
{"status": "On", "brightness": 2, "color": "20,7,0"}
The result was that the flash of white before the dull salmon color has gone.  That is what I wanted.
However, the initial warm white followed by the experimentally deliberate flash of cold white is now followed by a flash of dull salmon color, because my test setup allows me to loop through the test sequence.  Maybe this is something I will just have to live with and maybe there may be no way to actually set both the RGB and CCT modes to zero brightness simultaneously?  Is there no combination of command settings which would enable both the color and the color temperature to be set to zero brightness in the one "status": "Off" command?
My questions seem to relate to the behavior of the lights rather than the behavior of the ML Hub but since the light does not have a publically specified interface (API) I do greatly appreciate your help.

sidoh, 2020-05-07:
Happy to help navigate Milight's goofy protocol.

These facts probably aid quite a bit in understanding:

* The underlying protocol can only issue one command per packet (e.g., "set brightness to 0")
* As you've no doubt noticed, brightness values for color and white modes are independent
* The same command is used to change brightness for both color and white modes. It changes the brightness for whatever mode the bulb is in

All this together means that if you'd like to change the brightness for color mode while in white mode, you must first switch to color mode (where the bulb will have the previously set brightness value), and *then* set the brightness value.

So it's unfortunately impossible to avoid a flash of whatever the previous brightness value was.

You *can*, however, set the color or white value directly, as the way to tell the bulb to go into the appropriate mode is to set the color or whiteness value.

Hope this helps.

waparo, 2020-05-07:
That is indeed a goldmine of insights for me.
I'll set my usage within that framework.
I persevere with milights because the LED part is so good and comes as a 90mm ceiling down light.  I have built and tested a sonoff replacement for the goofy controller but producing and deploying umpteen of these with compliant power source is a lot of work and fairly expensive.
Chris,  thanks for your help.

sidoh, 2020-05-07:
Cool, sounds like that answers your question, so going to close.

closed 2020-05-07 by sidoh

sidoh, 2020-05-07:
btw - agreed re: Milight hardware. It's good stuff. Wish they'd have invested more in the software/protocol layer!