sidoh/esp8266_milight_hub #626: Problem with FUT018 bulb

UnchartedBull, 2020-05-03:
Hi, I've already consulted the troubleshooting guide, but I don't seem to find the issue. I have two FUT018, which I'd like to control. I ordered the nRF24L01+ from AliExpress and set everything up with Jumper Cables. After some trial and error I got the bulbs paired up and everything worked fairly smooth. Although after a power loss the pairing usually had to be redone.

Now I moved the ESP and nRF into a 3D printed case. I removed all the pins to make it fit and soldered the cables directly. I triple checked the wiring, but neither of my bulb is responding. So now it doesn't work anymore. I don't think, that I've damaged the components while depinning them. Website is loading up normal as well.

Now the weird part:
If I start sniffing packets there is of course nothing at first. Once I turn on a bulb I can see multiple packets received, which seem to come in a regular interval. I don't get any sent packets, even if I send a command to the bulb - is this normal? Also the ID in the received package is always 0001 (same id as the hub id they were paired to before with the jumper), even if I send a pair request for id 0x0002

So is it just a bad nRF module or is there something else I can try?

### What is the model number of the device you're trying to control?
FUT018

### What firmware version(s) have you tried? 
1.10.6 and 1.10.5

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 
NodeMCU

### Which radio type are you using? (RGBW, RGB+CCT, etc.) 
RGBW & RGB+CCT

### Have you tried controlling the device with a physical remote?
No

 

### Output of http://milight-hub.local/about and http://milight-hub.local/settings 

#### /about

```json
{"firmware":"milight-hub","version":"1.10.6","ip_address":"192.168.50.253","reset_reason":"External System","variant":"nodemcuv2","free_heap":12168,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### /settings

```json
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"--","mqtt_username":"milight","mqtt_password":"omitted","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/update/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Off","led_mode_packet":"Slow toggle","led_mode_packet_count":2,"hostname":"milight-hub","rf24_power_level":"HIGH","rf24_listen_channel":"MID","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"n","default_transition_period":500,"rf24_channels":["MID","HIGH"],"device_ids":[2],"gateway_configs":[],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{}}
```

sidoh, 2020-05-06:
If you're not seeing commands show up in 'sent' as you issue commands, it almost always means that the ESP8266 is not able to successfully communicate with the nRF24.

Make sure that the pins in the settings match how you've wired the nRF24. If that all looks correct, and the same hardware was working before, is it possible something was damaged?

UnchartedBull, 2020-05-07:
From the looks of it and from what I can measure everything looks fine. I've ordered a new one just to be sure though ... Will update this once the new one arrives.

sidoh, 2020-05-07:
Worth checking on serial logs during operation as well. (For this, I use `platformio device monitor`, but the Arduino IDE works as well if that's easier)

UnchartedBull, 2020-05-11:
The new should arrive today or tomorrow. I'll make sure to report back :)

UnchartedBull, 2020-05-19:
Just soldered in the new one and I still can't pair the bulbs. Something weird I noticed while sniffing:

The packet received commands seem to match exactly to the ones I sent, only weird thing is, that the ID won't switch if select another one from the dropdown. So for example if I load the webpage while on ID 0x2 and sent some packets, the traffic shows the packets as received with the correct ID. If I switch over to ID 0x1 and sent more packets the ID will stay the same (0x2) in the sniffed traffic. If I restart everything the ID correctly changes to 0x1, however then it will stay at 0x1 forever ...

UnchartedBull, 2020-05-19:
I did try the serial monitor and this might be interesting:

```
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.50.253
Wifi connected succesfully

Setup complete (version 1.10.6)
Unhandled websocket event: 10
```

Did you experience the websocket error before?

mozzhead164, 2020-06-08:
Bump.

Also getting the;
```
Setup complete (version 1.10.6)
Unhandled websocket event: 10
```
when connecting to milight-hub.local

I then cant send / receive any command as the LED stays on solid for a few seconds 
-- I have the 'LED mode when failed to connect' set to ON (default)
But there is nothing else printed in the debug window

is there any way to increase the level of debugging so i can dig deeper?

sidoh, 2020-06-08:
The websocket error is irrelevant -- just means client disconnect isn't being handled, which doesn't matter (there's no state to clean up on the server side).

> The packet received commands seem to match exactly to the ones I sent, only weird thing is, that the ID won't switch if select another one from the dropdown. 

I can't reproduce this. Can you upload a video?

> It has been working since then, but I've had the lot switched off for about a week, now getting very intermittent results. Could likely be a faulty / intermittent / counterfeit RF24 module

Intermittent failures like this have, in my experience, been one of the following:

* Poor SPI connection to nRF24
* Bad nRF24 module
* 2.4 GHz interference. Having some space between the ESP8266 and nRF24 seems to help

mozzhead164, 2020-06-16:
Thanks for the reply 

Managed to get it working it turned out to be a faulty breadboard connector. Was making connection sometimes and other times not. Hard to diagnose as it just looked like the ESP freezing up and unable to send data...

Worked through the troubleshooting guide and tested for continuity on each cable. Turned out to be either of the MISO or MIMO connectors (can’t remember exactly which)

Thanks for the support I’m now a happy user :) thanks for this nice project

sidoh, 2020-06-16:
Glad you were able to get it working! :)

@UnchartedBull, any luck? I'm guessing it's a connection issue. Happy to help if you've got further logs, data, etc. :)

UnchartedBull, 2020-06-18:
Finally got around to do some more testing (plus a Screen Recording). I turned on the LED on the ESP for "normal working mode" this time. After the ESP booted the LED continued to blink happily (just as configured). Once I hit the pair button it went completely dead though. I have fast blip configured for packages, but the LED was just off. After 1 or 2 minutes it came back on.

Also the sniffing is veeery delayed (around 20 seconds). I guess another option would be to completely wipe and reflash the ESP (haven't found an option to do that though ...). I might also add here, that I have pretty restricted firewalls and the ESP is sitting in the IoT VLAN (they are allowed to talk back to private network, but only on established connections). So I don't think that this is part of the issue.

Video is kinda lengthy, sorry. But I wanted to show everything. Let me know if I should try something else. I uploaded it as a normal video so you can skip ahead and pause :)

[Screen Recording 2020-06-18 at 17.55.03.mov.zip](https://github.com/sidoh/esp8266_milight_hub/files/4799711/Screen.Recording.2020-06-18.at.17.55.03.mov.zip)

sidoh, 2020-06-18:
By "connection," I mean the SPI connection between the nRF24 and ESP8266.

All of the symptoms you're highlighting are consistent with this. 

For example, if the LED stops flashing, it means the firmware isn't running the LED driver's `loop`, which means that it's stuck doing something else. The only place I've seen where this happens indefinitely is when the nRF driver is spin-looping waiting for a response over the SPI bus.

Double check:

* The SPI connection between ESP8266 and nRF24 is solid
* That the connection pinout matches your configurations in the Web UI
* That your nRF module works (easiest way to test this is with other firmware)

You can erase the ESP using `esptool.py`. Not something that can be done at the application level, as far as I'm aware.

UnchartedBull, 2020-06-19:
It's finally working! 🎉  I switched the modules around and all the cables were correct and correctly configured in the UI and it still wasn't working. The problem was that a super tiny wire (one of the strings inside a normal AWG wire) was touching D0 and D1. Since I use neither of those pins I didn't bother to test them. This wire probably found it's way there during the depinning of the nodeMCU.

Once I almost gave up I tested every pin and D0 and D1 didn't show continuity, although a very low resistance. I at first thought this was normal. Before I binned the board though, I had one close look again and noticed the small little wire between D0 and D1. After removing it everything works again. Thanks again for your help and patience! Truly an awesome project.

Maybe this can be added to the troubleshooting guide as a last resort to also test all other pins on the ESP, even if they aren't used (especially if the mcus are depinned to fit into a case or similar).

closed 2020-06-19 by UnchartedBull

sidoh, 2020-06-20:
Glad you were able to get it working! Thank you for following up with the fix. I added a bullet point to the troubleshooting guide.