sidoh/esp8266_milight_hub #627: Problem setting colour from mqtt (home assistant) on a RGB+CCT bulb (protocol 6)

Montie001, 2020-05-05:
### Describe the bug

I have a few RGB_CCT milight bulbs, If I use the web interface for the hub I can control ALMOST normally. MQTT I cant.

### Steps to reproduce

I can click several colours and the bulb changes as expected.

If I then click say a colour  temperature it will change to the white as expected then flash back to the last selected colour - sometimes back to white then back to the colour again.

If I click white button it will stay white.

My main problem is that I cannot set the colour from home assistant.  I can set the colour temperature. If I select a colour It changes to the chosen colour, cycles to previously set colour (set from the web interface) then goes back to white. 

I checked the mqtt messages and they are normal- the extraneous changes aren't in the mqtt commands.

I can choose a effects mode ok and white mode ok.

My RGB bulbs are working fully normally.

### Setup information

#### Firmware version

Current Version
1.10.6 (nodemcuv2)
Latest Version
Version
1.10.6
Release Date
2020-04-18T21:10:01Z

#### Output of http://milight-hub.local/about
{"firmware":"milight-hub","version":"1.10.6","ip_address":"192.168.1.46","reset_reason":"External System","variant":"nodemcuv2","free_heap":148{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":125,"http_repeat_factor":5,"auto_restart_period":0,"mqtt_server":"192.168.1.160:1883","mqtt_username":"xxi","mqtt_password":"xx","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/update/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/state/:device_id/:device_type/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":0,"state_flush_interval":750,"mqtt_state_rate_limit":1000,"mqtt_debounce_delay":500,"packet_repeat_throttle_sensitivity":10,"packet_repeat_throttle_threshold":300,"packet_repeat_minimum":3,"enable_automatic_mode_switching":true,"led_mode_wifi_config":"Slow toggle","led_mode_wifi_failed":"Fast toggle","led_mode_operating":"Off","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"HIGH","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","HIGH"],"device_ids":[43793,43795,43797],"gateway_configs":[[43793,8801,5],[43794,8802,5],[43795,8803,5],[43796,8804,5],[43797,8805,6],[43798,8806,5],[43799,8807,6],[43800,8808,6],[43801,8809,5]],"group_state_fields":["state","status","brightness","color","mode","color_temp","bulb_mode","effect"],"group_id_aliases":{}}56,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}

sidoh, 2020-05-06:
Set `enable_automatic_mode_switching` to false. That's what this feature does (it's disabled by default).

sidoh, 2020-05-06:
Let me know if I've misunderstood and we can re-open.

closed 2020-05-06 by sidoh