sidoh/esp8266_milight_hub #629: White mode not working properly on MQTT with FUT037

Wronek, 2020-05-06:
In web interface of MiLight Hub i can set propellery white mode or night mode. 

I tried send a state manualy to mqtt topic using app on my mobile, and there command like "toggle" or "next_mode" work propellery, but commond "set_white" not working. I tried set a white mode, also as effect, but this also not working.
I noticed also hub not set all of three color together. For example if i sens a red:100 green: 150 blue: 250 hub not set this color because all color are used. Though when i set red:0 green:150 blue:250 hub set this color.

Current Version
1.10.6 (nodemcuv2)
[image]
[image]
[image]

sidoh, 2020-05-06:
Sorry, I'm not following. Can you please include the full commands you're trying, and what you expect to happen with each of them?

Wronek, 2020-05-07:
{"command":"set_white"} or {"effect":"white_mode"} i expect white color of my led strip but color of strip not changed

In HA white, also not working.  

In HA i see second problem with colors. Please see video. When a setting color the point come back outside circle because hub not seting mix three colors.  Setting only colors where use only two colors of rgb. 
https://youtu.be/Y76IFx9oudU

sidoh, 2020-05-07:
It looks like you're using the RGBW remote type, which doesn't support saturation. That limits the colors to the ones on the outer wheel. While this is confusing, it's sort of a limitation of HomeAssistant's UI.

sidoh, 2020-05-07:
For the first issue (where the `set_white` command isn't being respected), can you post a packet capture log while sending the commands? (This is the output in the Web UI after you click on "Start Listening").

My guess is a packet is being sent, but your device isn't recognizing it for some reason.

Do you have an official remote that works as expected? If so, post a packet capture log while setting color to white.

Wronek, 2020-05-08:
Thanks for the answer.

What can I do to get white in HA?

Setting white using pilot from image below.
[image]
[image]

When i sending set_white from my phone app ,,Traffic sniffed" nothing display. But when i send toggle work very well...
See image below for toggle command :
[image]

Thanks for your time