sidoh/esp8266_milight_hub #630: States keep updating specific device ID

poudenes, 2020-05-07:
### Describe the bug

I saw in MQTT Explorer that the hub sends a bulk states messages from 1 specific device id. (0xAAAF)

### Steps to reproduce

Totaly unclear how to reproduce it

### Setup information
Wemos D1 R2 Board
Home Assistant

#### Firmware version
1.10.5 (nodemcuv2)

#### Output of http://milight-hub.local/about

```
{"firmware":"milight-hub","version":"1.10.5","ip_address":"192.168.100.70","reset_reason":"Software/System restart","variant":"nodemcuv2","free_heap":14360,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```
{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.100.24:1883","mqtt_username":"PRIVATE","mqtt_password":"PRIVATE","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"milight/states/:hex_device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":200,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":10,"packet_repeat_throttle_threshold":500,"packet_repeat_minimum":50,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"Fast blip","led_mode_operating":"Slow toggle","led_mode_packet":"Fast blip","led_mode_packet_count":4,"hostname":"MilightHub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"192.68.100.70","wifi_static_ip_gateway":"192.168.100.X","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":50,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","HIGH"],"device_ids":[43681,43682,43683,43684,43690,43691,43692,43693,43694,43695,43706,43707],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type"],"group_id_aliases":{}}
```

### Additional context

sidoh, 2020-05-07:
What's going on on the hub side? If you enable packet sniffing, to you see packets going out? Any messages on the command topic?

poudenes, 2020-05-08:
I will try to keep eye on it. Yesterday it was just a coincidence i saw it happen.
when i see it happen i'll start sniffing

poudenes, 2020-05-13:
I don't see them anymore.

closed 2020-05-13 by poudenes