sidoh/esp8266_milight_hub #631: Retain

poudenes, 2020-05-13:
Hi All,

I had a question about follow. In Wiki I see how you can add lights manual in HA.
I don't see the option "retain". I add this myself in the config for all my lights:

```
###########################################################################################
#
# MQTT LIGHT
#
# https://www.home-assistant.io/integrations/light.mqtt/
#
###########################################################################################
---
platform: mqtt
name: "Sink 1"
schema: json
command_topic: "milight/0xAAA1/rgb_cct/1"
state_topic: "milight/states/0xAAA1/rgb_cct/1"
qos: 1
optimistic: "false"
retain: "true"
brightness: "true"
rgb: "true"
color_temp: "true"
effect: "true"
effect_list: !include /config/package/templates/light_effects.yaml
availability_topic: "milight/client_status"
payload_available: "connected"
payload_not_available: "disconnected"
```

Is it ok to do this or is there a reason to not to do this?

When I restart HA then lights go on by themselves. Even I had them off in HA before restart

agoralive, 2020-05-18:
May I suggest that you clear/erase you MQTT broker db.

What setup are you using for your broker?

poudenes, 2020-05-18:
Mosquitto Broker running in Docker
HA Add-On

I cleared already dB because of some issues. 
Last night the Milight Hub lost wifi connection. After reconnect the bedroom light went on at 5am
11 minutes later we woke up because of the light.

agoralive, 2020-05-18:
Try setting 

`retain: false`

In you configuration file for the specific card

poudenes, 2020-05-18:
> Try setting
> 
> `retain: false`
> 
> In you configuration file for the specific card

Already have that in my config. 
But the Hub also sends MQTT out with retain.

agoralive, 2020-05-18:
Multiple hubs?

poudenes, 2020-05-18:
1 Wemos D1R2 board with latest Firmware 1.10.6
Home Assistant with MQTT Add-On at this moment

Also had MQTT on its own RPI Zero. Same issues

My Bulbs in total:
**Sink**
```
43681 	0xAAA1/rgb_cct/0 - sk0
	0xAAA1/rgb_cct/1 - sk1
	0xAAA1/rgb_cct/2 - sk2
	0xAAA1/rgb_cct/3 - sk3
```
**Kitchen**
```
43682 	0xAAA2/rgb_cct/0 - kr0
	0xAAA2/rgb_cct/1 - kt1
	0xAAA2/rgb_cct/2 - kt2
	0xAAA2/rgb_cct/3 - kt3
```
**Hallway**
```
43683 	0xAAA3/rgb_cct/1 - hl
```
**Balcony**
```
43684	0xAAA4/rgb_cct/1 - bc
```
**Bedroom**
```
43706 	0xAABA/rgb_cct/1 - br0 (contain all 6 bulbs)
43690	0xAAAA/rgb_cct/1 - br1
	0xAAAA/rgb_cct/2 - br2
	0xAAAA/rgb_cct/4 - br3
	0xAAAA/rgb_cct/3 - br4

43691	0xAAAB/rgb_cct/1 - br5
	0xAAAB/rgb_cct/2 - br6
```
**Livingroom**
```
43692 	0xAAAC/rgb_cct/0 -lr0
	0xAAAC/rgb_cct/1 - lr1
	0xAAAC/rgb_cct/2 - lr2
	0xAAAC/rgb_cct/3 - lr3
	0xAAAC/rgb_cct/4 - lr4
```
**Bathroom**
```
43693 	0xAAAD/rgb_cct/0 - bt0
	0xAAAD/rgb_cct/1 - bt1
	0xAAAD/rgb_cct/2 - bt2
	0xAAAD/rgb_cct/3 - bt3
	0xAAAD/rgb_cct/4 - bt4
```
**Toilet**
```
43694 	0xAAAE/rgb_cct/0 - wc0
	0xAAAE/rgb_cct/1 - wc1
	0xAAAE/rgb_cct/2 - wc2
	0xAAAE/rgb_cct/3 - wc3
```
**Dressoir**
```
43695 	0xAAAF/rgb_cct/1 - dr0
	0xAAAF/rgb_cct/1 - dr1
	0xAAAF/rgb_cct/2 - dr2
	0xAAAF/rgb_cct/3 - dr3
```
**Wall and Window**
```
43707	0xAABB/rgb_cct/1 - wall (contain 2 bulbs)
	0xAABB/rgb_cct/2 - window (contain 2 bulbs)
```
**MQTT TOPICS**
```
	"mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
	"mqtt_update_topic_pattern": "milight/updates/:hex_device_id/:device_type/:group_id",
	"mqtt_state_topic_pattern": "milight/states/:hex_device_id/:device_type/:group_id",
	"mqtt_client_status_topic": "milight/client_status",
```
Here a example of my MQTT Light config:

```
###########################################################################################
# MQTT LIGHT
# https://www.home-assistant.io/integrations/light.mqtt/
###########################################################################################
---
platform: mqtt
name: "Balcony"
schema: json
command_topic: "milight/0xAAA4/rgb_cct/1"
state_topic: "milight/states/0xAAA4/rgb_cct/1"
qos: 1
optimistic: "false"
retain: "false"
brightness: "true"
rgb: "true"
color_temp: "true"
effect: "true"
effect_list: !include /config/package/templates/light_effects.yaml
availability_topic: "milight/client_status"
payload_available: "connected"
payload_not_available: "disconnected"
```

jjok, 2020-05-26:
These should be boolean, rather than strings:
```
optimistic: false
retain: false
brightness: true
rgb: true
color_temp: true
```
I assume `"false"` is truthy in YAML.

poudenes, 2020-05-27:
> These should be boolean, rather than strings:
> 
> ```
> optimistic: false
> retain: false
> brightness: true
> rgb: true
> color_temp: true
> ```
> 
> I assume `"false"` is truthy in YAML.

This is in YAML. Everywhere i have "false" and "true" between "" HA is running perfect.
I thing its something with retain. Hub is sending topics also with retain.

When i restart the Wemos i see some lights turning on, even when they are off in MQTT topics.
Maybe something inside the HUB software. Last night my livingroom lights went on for 30min and off again. (I was a sleep and didnt turn them off)

sidoh, 2020-11-01:
Sorry I missed this.

I'm not sure I'm following what's happening here. If you turned your lights off, the retained message should reflect that, right? I don't think the hub reconnecting should set bulbs to a different status than they were previously unless state gets out of sync.

sidoh, 2020-11-03:
Adding a setting in 1.10.7. Live in the 1.10.7 branch, but going to get a few more changes in before cutting a release.

closed 2020-11-25 by sidoh