sidoh/esp8266_milight_hub #632: Device names all correct in WebUI but some missing in settings.json file

poudenes, 2020-05-15:
Hi All,

I add for all devices a device name. I'll checked in WebUI and all devices have the given name.
When I export the settings there are devices missing (See JSON Below).

I don't know if this is a bug, or something else.

```
Sink
43681 	0xAAA1/rgb_cct/0 - sk0
		0xAAA1/rgb_cct/1 - sk1
		0xAAA1/rgb_cct/2 - sk2
		0xAAA1/rgb_cct/3 - sk3

Kitchen
43682 	0xAAA2/rgb_cct/0 - kr0
		0xAAA2/rgb_cct/1 - kt1
		0xAAA2/rgb_cct/2 - kt2
		0xAAA2/rgb_cct/3 - kt3

Hallway
43683 	0xAAA3/rgb_cct/1 - hl

Balcony
43684	0xAAA4/rgb_cct/1 - bc

Bedroom
43706 	0xAABA/rgb_cct/1 - br0 (contain all 6 bulbs)

43690	0xAAAA/rgb_cct/1 - br1
		0xAAAA/rgb_cct/2 - br2
		0xAAAA/rgb_cct/4 - br3
		0xAAAA/rgb_cct/3 - br4
43691	0xAAAB/rgb_cct/1 - br5
		0xAAAB/rgb_cct/2 - br6

Livingroom
43692 	0xAAAC/rgb_cct/0 -lr0
		0xAAAC/rgb_cct/1 - lr1
		0xAAAC/rgb_cct/2 - lr2
		0xAAAC/rgb_cct/3 - lr3
		0xAAAC/rgb_cct/4 - lr4

Bathroom
43693 	0xAAAD/rgb_cct/0 - bt0
		0xAAAD/rgb_cct/1 - bt1
		0xAAAD/rgb_cct/2 - bt2
		0xAAAD/rgb_cct/3 - bt3
		0xAAAD/rgb_cct/4 - bt4

Toilet
43694 	0xAAAE/rgb_cct/0 - wc0
		0xAAAE/rgb_cct/1 - wc1
		0xAAAE/rgb_cct/2 - wc2
		0xAAAE/rgb_cct/3 - wc3

Dressoir
43695 	0xAAAF/rgb_cct/1 - dr0
		0xAAAF/rgb_cct/1 - dr1
		0xAAAF/rgb_cct/2 - dr2
		0xAAAF/rgb_cct/3 - dr3

Wall and Window
43707	0xAABB/rgb_cct/1 - wall (contain 2 bulbs)
		0xAABB/rgb_cct/2 - window (contain 2 bulbs)
```

```
{
	"admin_username": "",
	"admin_password": "",
	"ce_pin": 16,
	"csn_pin": 15,
	"reset_pin": 0,
	"led_pin": -2,
	"radio_interface_type": "nRF24",
	"packet_repeats": 50,
	"http_repeat_factor": 1,
	"auto_restart_period": 0,
	"mqtt_server": "192.168.100.XXX:1883",
	"mqtt_username": "XXXXX",
	"mqtt_password": "XXXXX",
	"mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
	"mqtt_update_topic_pattern": "milight/updates/:hex_device_id/:device_type/:group_id",
	"mqtt_state_topic_pattern": "milight/states/:hex_device_id/:device_type/:group_id",
	"mqtt_client_status_topic": "milight/client_status",
	"simple_mqtt_client_status": true,
	"discovery_port": 48899,
	"listen_repeats": 3,
	"state_flush_interval": 200,
	"mqtt_state_rate_limit": 500,
	"mqtt_debounce_delay": 500,
	"packet_repeat_throttle_sensitivity": 10,
	"packet_repeat_throttle_threshold": 500,
	"packet_repeat_minimum": 50,
	"enable_automatic_mode_switching": false,
	"led_mode_wifi_config": "Fast toggle",
	"led_mode_wifi_failed": "Fast blip",
	"led_mode_operating": "Slow toggle",
	"led_mode_packet": "Fast blip",
	"led_mode_packet_count": 4,
	"hostname": "MilightHub",
	"rf24_power_level": "MAX",
	"rf24_listen_channel": "LOW",
	"wifi_static_ip": "192.68.100.xxx",
	"wifi_static_ip_gateway": "192.168.100.1",
	"wifi_static_ip_netmask": "255.255.255.0",
	"packet_repeats_per_loop": 50,
	"home_assistant_discovery_prefix": "",
	"wifi_mode": "g",
	"default_transition_period": 500,
	"rf24_channels": ["LOW", "MID", "HIGH"],
	"device_ids": [43681, 43682, 43683, 43684, 43690, 43691, 43692, 43693, 43694, 43695, 43706, 43707],
	"gateway_configs": [],
	"group_state_fields": ["state", "status", "brightness", "level", "hue", "saturation", "color", "mode", "kelvin", "color_temp", "bulb_mode", "computed_color", "effect", "device_id", "group_id", "device_type"],
	"group_id_aliases": {
		"bc": ["rgb_cct", 43684, 1],
		"br1": ["rgb_cct", 43690, 1],
		"br2": ["rgb_cct", 43690, 2],
		"br3": ["rgb_cct", 43690, 3],
		"br4": ["rgb_cct", 43690, 4],
		"br5": ["rgb_cct", 43691, 1],
		"br6": ["rgb_cct", 43691, 2],
		"bt0": ["rgb_cct", 43693, 0],
		"bt4": ["unknown", 0, 0],
		"ds0": ["rgb_cct", 43695, 0],
		"ds1": ["rgb_cct", 43695, 1],
		"ds2": ["rgb_cct", 43695, 2],
		"ds3": ["rgb_cct", 43695, 3],
		"hl": ["rgb_cct", 43683, 1],
		"kc0": ["rgb_cct", 43682, 0],
		"kc1": ["rgb_cct", 43682, 1],
		"kc2": ["rgb_cct", 43682, 2],
		"kc3": ["rgb_cct", 43682, 3],
		"lr0": ["rgb_cct", 43692, 0],
		"lr1": ["rgb_cct", 43692, 1],
		"lr2": ["rgb_cct", 43692, 2],
		"lr3": ["rgb_cct", 43692, 3],
		"lr4": ["rgb_cct", 43692, 4],
		"sk0": ["rgb_cct", 43681, 0],
		"sk1": ["rgb_cct", 43681, 1],
		"sk2": ["rgb_cct", 43681, 2],
		"sk3": ["rgb_cct", 43681, 3]
	}
}
```

poudenes, 2020-05-17:
After refresh of browser some devices are now also missing name in the WebGui.

sidoh, 2020-05-29:
There's a limit on the size of the settings document (I believe it's 2KiB), which functionally means there's a limit on the number of device aliases. 

Think this is a duplicate of #569.

closed 2020-05-29 by sidoh