sidoh/esp8266_milight_hub #634: WebUI "broken"

Housefrau, 2020-05-27:
Hello together!

Hope you are doing all well. 

Since a few weeks (Unfortunately I cannot determine an exactly date) I am not able to get the milight-hub working properly on my nodemcuv3 CH340G. It worked like a charm for years and somewhen the WebUI just looks like in the screenshot. I tried to flash different versions, also flashed a 1MB blank file between the version switches -> No changes. :(

I flashed the firmware with the NodeMCU-Flasher for Windows (suggested in the blog) and also with PlatformIO on Unix. Both results in the "broken" WebUI.

I guess this is something I can fix myself, but honestly I just don't know how. Hopefully, someone is out there to help me out. :-)

Just let me know, if you need any further informations.

[image]

### What is the model number of the device you're trying to control?
FUT018

### What firmware version(s) have you tried? 
1.10.6
1.10.5
1.9.2

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 
nodemcuv3

### Which radio type are you using? (RGBW, RGB+CCT, etc.) 
RGBW

### Have you tried controlling the device with a physical remote?
no

### Output of http://milight-hub.local/about and http://milight-hub.local/settings 
#### /about

```json
{"firmware":"milight-hub","version":"1.9.2","ip_address":"192.168.201.11","reset_reason":"Software/System restart","variant":"nodemcuv2","free_heap":22328,"arduino_version":"2_4_2"}
```

#### /settings
I recieve a timeout, when try to open /settings.

Tontze, 2020-05-27:
Did you erase before flash?

Housefrau, 2020-05-27:
> 
> 
> Did you erase before flash?

Yes I did. Could that be an hardware issue?

Housefrau, 2020-05-27:
Never mind - I just found it. Damn.

I am using Pi-Hole in my network and it seems like the URL "maxcdn.bootstrapcdn.com" is requested, which has been blocked.

URL whitelisted -> Everything's fine.

Thanks for your time!

closed 2020-05-27 by Housefrau