sidoh/esp8266_milight_hub #635: Fut087 recognized as fut089

Tontze, 2020-05-27:
Dont know does it really matter, but just received my [FUT087](https://www.futlight.com/productdetails.aspx?id=341&typeid=143), and it is recognized as FUT089. Attleast i see mqtt messages as [FUT089](https://www.futlight.com/productdetails.aspx?id=276&typeid=143).

sidoh, 2020-05-29:
It probably uses the same protocol. Unless it's causing any noticeable issues, this should be fine. Let me know if that's the case.

closed 2020-05-29 by sidoh

kodifan, 2021-12-08:
hallo sidoh 
I know this issue is closed but i have noticed the same thing. FUT087 is recognized as a FUT089. It basically do not causing any major issues but those are completely different remotes. FUT087 is a mono (CCT only remote with just one group), FUT089 is a RGBW, 8 group remote. And my HA recognize all those capabilities incorrectly . Is there a much work to implement proper recognizing of this remote ? If its not it will be great to see this added, if it will take to much time and effort please ignore this :)