sidoh/esp8266_milight_hub #637: Add dimming features

etsach, 2020-06-03:
Most connected wall switches have actions for starting / stopping dimming, such as long press to start dimming, and short press to stop. It would be interesting to have this enabled via MQTT. Milight protocol does not have these commands but they can be easily emulated as lamps react very fast.
With home automation  systems such as domoticz, scripting is way too slow for emulating this features via scripts or plugins, brightness could be changed only every few seconds.

These four commands would be very useful :
 - Dim up (automatically stops when reaching max brightness)
 - Dim down (automatically stops when reaching min brightness without switching off)
 - Dim up and down (sweeps brightness up and down, stopping after something like 10 cycles)
 - Stop dimming

sidoh, 2020-06-07:
Not sure if this is exactly what you're talking about, but transitions between any two bulb states are already supported:

https://github.com/sidoh/esp8266_milight_hub#transitions

Not all of your examples are supported, but any simple linear transition works.

etsach, 2020-06-08:
Ah yes, I forgot about transitions, but you still can't send a command for
stopping the transition to its current state.

The idea behind this is for implementing dimming features with conventional
push buttons, emulating conventional dimmers. This is a convenient solution
if you want to keep traditional push buttons in the home (for design &
style, you may not want to add modern stuff like touch controls) and this
behavior has a good acceptance factor by non geeks.

With two way buttons, you will send a transition up or down command on a
press&hold, and a stop command on release. We are just missing the stop
command here.

With a single press button (very common on conventional dimmers), you start
cycling on a long press, and stop at desired level on a short press. Here
we miss the cycling transition command and stop command.

So we would need:
 - A "stop&freeze transition" command
 - A transition option for cycling between current and specified states,
instead of just stopping at specified state

Etienne

sidoh, 2020-06-08:
What is the purpose of cycling between states? This is not well-supported by the current architecture, and unless there's pretty substantial utility, I don't think the added complexity is justified.

The REST API has a cancel command (`DELETE /transitions/:id`). See [docs for further detail](https://sidoh.github.io/esp8266_milight_hub/branches/latest/#tag/Transitions/paths/~1transitions~1{id}/delete). Probably at least slightly awkward to accomplish what you're describing, but the support is there.

etsach, 2020-06-28:
The purpose is to allow setting up dimming with a classical hardware wall switch. As wall switches have no native dimming input (such as touch slider, or knob), what is usually done is : a first input enables a dimming state, for instance a slow brightness ramp, and a second input stops the brightness ramp to the current desired value.
To emulate this with a home controller script,  we would need at least a command that stops the current ongoing transition.

poudenes, 2020-07-05:
Does someone know if you dim a bulb from the App (Milight) it will send 1 command to dim and the bulb do the rest, or will it full controlled by the HUB? If the bulb control the dimming it would be nice to have this implement.

When you send a dim command from HA or a transition to the bulb, and the bulb do the cycle you can make free some things in the HUB. HUB Is free to accept new commands. Now when you use transition the HUB is busy to control the transition and other command are waiting in a queue.

sidoh, 2020-11-01:
Sorry for the very late reply.

Milight bulbs do not natively support any kind of transitions.

There is a command to delete transitions. You _can_ cancel transitions -- the `/transitions` API routes are RESTful. :)

Going to close this, but happy to continue discussion.

closed 2020-11-01 by sidoh