sidoh/esp8266_milight_hub #638: Home Assistant MQTT auto discovery

TroLoos, 2020-06-04:
Hi, auto discovery actually never worked for me - I've set up all my lights manually in config files. But through time I started to use different kinds of auto discovery in Home Assistant and they worked great... espMH is the only one left in manual mode, can you help me change it?

In my observation - what auto discovery publishes in homeassistant topic is one level too deep. In my case it is:
homeassistant/light/milight_hub_xxxxxxx/(in here multiple lights)

All my other auto-discovered devices are defined straight after homeassistant/light/ or any other type of component.

The end result is "Unnamed device" in Home Assistant with no child devices whatsoever. What am I missing here? I'm sure the problem is very close to the surface :-)

sidoh, 2020-06-04:
They may have changed something in more recent versions. I've tested it as recently as Jan this year and it works.

The optional [node_id] component of the MQTT path appears to still be a thing according to the docs:

https://www.home-assistant.io/docs/mqtt/discovery/

Not near my hub right now, but can test later.

TroLoos, 2020-06-05:
Ok, so I have no idea why it doesn't work for me then...

It's been like this for last 20+ Home Assistant versions. My espMG version is also the newest (1.10.6 (nodemcuv2) and has been like this for last 2+ versions.

[image]

sidoh, 2020-06-05:
I'm sure it's a problem, happy to take a look. Should have time today.

Could you post the output of `/settings` and a full MQTT message from the auto-discovery topic as well?

sidoh, 2020-06-05:
(When posting `/settings` output, make sure to sensor any passwords)

TroLoos, 2020-06-05:
Excellent!

Here's my /settings page (I'm sorry but I am not sure how to format this input properly):
`{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"10.0.20.2","mqtt_username":"xxxxxxxx","mqtt_password":"xxxxxxxx","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"10.0.20.100","wifi_static_ip_gateway":"10.0.20.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[26548,28790],"gateway_configs":[],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{"Bright_Bathroom":["rgb_cct",26548,4],"Bright_Dinner_Table":["rgb_cct",26548,2],"Bright_Lounge":["rgb_cct",26548,1],"Bright_Master_Bedroom":["rgb_cct",26548,3],"Color_Bathroom":["rgb_cct",28790,4],"Color_Dinner_Table":["rgb_cct",28790,2],"Color_Lounge":["rgb_cct",28790,1],"Color_Master_Bedroom":["rgb_cct",28790,3]}}`

and following one of the auto-discovery topics:
>{
>  "schema": "json",
>  "name": "Bright_Bathroom",
>  "command_topic": "milight/0x67B4/rgb_cct/4",
>  "state_topic": "milight/states/0x67B4/rgb_cct/4",
>  "device": {
>    "manufacturer": "esp8266_milight_hub",
>    "sw_version": "1.10.6",
>    "identifiers": [
>      9066451,
>      26548,
>      "rgb_cct",
>      4
>    ]
>  },
>  "availability_topic": "milight/client_status",
>  "payload_available": "connected",
>  "payload_not_available": "disconnected",
>  "brightness": true,
>  "effect": true,
>  "effect_list": [
>    "night_mode",
>    "white_mode",
>    "0",
>    "1",
>    "2",
>    "3",
>    "4",
>    "5",
>    "6",
>    "7",
>    "8"
>  ],
>  "rgb": true,
>  "color_temp": true
>}

sidoh, 2020-06-07:
It appears to be working for me, here's my setup:

1. Latest stable release of Homeassistant (1.110.5) running in docker locally
2. Mosquitto MQTT broker also in docker
3. Test espMH connecting to same MQTT broker

I enabled MQTT in Homeassistant through the UI, making sure to check the "Enable discovery" box.

How do you have MQTT configured within Homeassistant?

screenshots:

[image]

[image]

TroLoos, 2020-06-09:
Ok, so I understand, that there's nothing wrong with my config, and the reason why it does not work is unknown, right?

I think my setup works fine as I have 30+ devices connected to Home Assistant via MQTT auto discovery feature (Tasmota ESP8266 wall plugs, zigbee2mqtt aqara switch, Shelly 2.5 / 1 / 1PM).

So I will keep my install as it is now then...

sidoh, 2020-06-09:
That is my understanding. Hard for me to poke at this further without being able to reproduce. Discovery has worked for others as well, for what it's worth.

This should be [getting handled](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MQTT/HomeAssistantDiscoveryClient.cpp#L134), but you can try using `homeassistant/` instead of `homeassistant` as the discovery prefix.

What software are you using for an MQTT broker?

Also worth making sure the MQTT messages that should be getting published are getting published (easiest way is to attach a CLI client like `mosquitto_sub` to `homeassistant/#`).

TroLoos, 2020-06-16:
I'm in the middle of setting up my fresh Home Assistant installation, based on HassOS. When trying to setup MQTT discovery, I caught the following error in logs. Perhaps it can help us isolate my problem:

Log Details (ERROR)
Logger: homeassistant.util.logging
Source: util/logging.py:141
First occurred: 10:37:42 (1 occurrences)
Last logged: 10:37:42

Exception in state_received when handling msg on 'milight/states/0x67B4/rgb_cct/2': '{"brightness":105,"color":{"r":255,"g":255,"b":255}}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 35, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 248, in state_received if values["state"] == "ON": KeyError: 'state'

sidoh, 2020-06-16:
It looks like the `state` key is missing, which is required by HomeAssistant.

Can you try turning that bulb on? (RGB+CCT, `0x67B4`, group 2).

I'm not sure exactly how it got in this state. Shouldn't be possible to set fields without first turning the bulb on. But here we are 😅

TroLoos, 2020-06-16:
It auto-added eventually :-) Strange... good to keep that in mind in case any further incompatibilities show up :-)

closed 2020-06-16 by TroLoos