sidoh/esp8266_milight_hub #639: Transition questions

markg85, 2020-06-07:
Hi,

I have 2 questions about transitions.

1. The web admin does allow me to set a transition period, but no duration. What is the default duration and how can i change it? This is in a situation where i use this project along with the smartthings integration from [here](https://github.com/fireboy1919/MiThings/tree/master/smartapps/fireboy1919). So going the MQTT route (where the duration can be set) is a not really an option. Changing the code from said project to allow a duration is possible though (not preferred). I much rather have a default duration which is what i'm looking for.

2. How do transitions work and how is the brightness "gamma corrected" from linear to exponential? Oke, these are two questions on their own :) I suppose what i'm looking for is how a transition is done. I did look at the code and from there i "think" the transition is updated by the given period in milliseconds and just increased. That would be (assumption) a linear increase. That alone would be wrong, but i suppose the LED bulbs themselves do the brightness correction converting a linear curve to an exponential curve? If you have no clue what i'm talking about here, read [this very helpful document](https://cdn-learn.adafruit.com/downloads/pdf/led-tricks-gamma-correction.pdf) about gamma correction.

Best regards,
Mark

sidoh, 2020-06-07:
Hi Mark,

The settings you see in the UI are the settings that exist. The duration is [hardcoded here](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/Transitions/Transition.h#L73). You can recompile with this constant changed if you like.

I'd welcome a PR to expose this as a setting, but is realistically not something I'm going to add myself in the near future. If you'd like to take a stab at it, #527 probably serves as a pretty good reference.

Re: (2), I'm happy to discuss, but my initial feeling is that non-linear transitions are way beyond the complexity:utility ratio that I'm comfortable with for this project. Ideally hardware would adjust for this (otherwise the brightness values are "wrong" to begin with), but knowing Milight, that's probably not the case.

markg85, 2020-06-07:
Hi Chris,

Is the transition part of the data packet send to the bulb? So that the bulb itself handles the transition? As if that is the case, the data packet likely has a few bits for the duration too.

I was asking about #2 because i don't see much of a transition. There definitely is a tiny transition but from what i'm seeing that's well within half a second which seems a little too fast for me. I'd like my lights to fade in fast enough to notice the fade but not slow enough that you're waiting for the fade to be completed. Somewhere between half a second and a second is fine (depending on the animation).

Note that gamma correction "sounds" complicated, but you have tools that generate the lookup table for you :) Then it's just a matter of mapping your steps (say 1-1024) to the (1-1024) values of the gamma correction table. But i'm actually assuming that the light bulbs themselves do the correction.

sidoh, 2020-06-07:
Milight packets are dead simple. They tell the bulb to change the value of a single setting (e.g., "set brightness to 100"). Transitions are entirely managed by espMH.

The default transition duration is 10,000ms. If it looks shorter than that, chances are the command isn't being issued properly (the [transitions section of the README](https://github.com/sidoh/esp8266_milight_hub#transitions) should be helpful). A command issued via MQTT does not result in a transition by default -- you need to include the `transition` key in the payload (the value is the duration). The MQTT payload schema is adopted from HomeAssistant.

The complexity I'm referring to is composed 95% of the plumbing and special casing needed to support this for brightness, but not other fields (like hue and color temperature) and 5% the mechanics of gamma correction.

Also worth considering that the proper LUT depends on how Milight is translating the brightness value into PWM parameters, which I have no way of knowing without tearing down a bulb and probing.

closed 2020-11-01 by sidoh