sidoh/esp8266_milight_hub #640: Flashing lights after one command

daniel9522, 2020-06-11:
I have just installed the latest version of the hub it's my first time using it but after turning on any lights it will start flashing the bulb or led strip
I tried playing around with the radio settings but no luck

`{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.1.4:1883","mqtt_username":"danielmartin","mqtt_password":"help124","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/commands/","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":1,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Slow blip","led_mode_packet_count":5,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":1,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","HIGH"],"device_ids":[1,2,3,4,5,6,7,8,9],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect"],"group_id_aliases":{"Hallway 1floor":["cct",9,1],"hallway 2floor":["cct",3,1],"kitchenambient":["rgbw",8,1],"livingroomambientsofa":["rgbw",6,1],"livingroomambientwindow":["rgbw",7,1],"masterbedroom":["rgbw",1,1],"masterbedroom night":["rgbw",2,1],"studyambient":["rgbw",4,1],"studypc":["rgbw",5,1]}}`

Any help would be appreciated
Thank you

sidoh, 2020-06-27:
Sorry, missed this.

If you could record a short video showing what you're doing in the UI and how the lights respond, that'd shortcut a bunch of questions.

Does this happen with all lights?

akawoj, 2020-06-28:
Well, I can provide mosquitto_sub output if it helps.

The setup is following:
1. MQTT on ESP is configured with "homeassistant" HomeAssistant MQTT Discovery Prefix and homeassistant discovers proper entities.

2. After adding the "light" card to lovelace with one of discovered milight hub related entities (doesn't matter which) it can be turned on and it lights up. After trying to set it off withe the lovelace it starts blinking. The only way to make it stop is setting wrong mqtt severd IP address on esp hub.

What I can see in mosquitto_sub - there is heavy traffic going in "blinking state":

`milight/0xEF3/cct/3 {"state":"OFF"}
milight/states/0xEF3/cct/3 {"state":"OFF","brightness":255,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/0xEF3/cct/3 {"state":"ON"}
milight/0xEF3/cct/3 {"state":"OFF"}
milight/0xEF3/cct/3 {"state":"ON"}
milight/states/0xEF3/cct/3 {"state":"ON","brightness":255,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/0xEF3/cct/3 {"state":"OFF"}
milight/0xEF3/cct/3 {"state":"ON"}
milight/states/0xEF3/cct/3 {"state":"ON","brightness":255,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/0xEF3/cct/3 {"state":"OFF"}
milight/0xEF3/cct/3 {"state":"ON"}
milight/states/0xEF3/cct/3 {"state":"ON","brightness":255,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/0xEF3/cct/3 {"state":"OFF"}
milight/0xEF3/cct/3 {"state":"ON"}
milight/states/0xEF3/cct/3 {"state":"ON","brightness":255,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/0xEF3/cct/3 {"state":"OFF"}
milight/0xEF3/cct/3 {"state":"ON"}
milight/states/0xEF3/cct/3 {"state":"ON","brightness":255,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/0xEF3/cct/3 {"state":"OFF"}
milight/0xEF3/cct/3 {"state":"ON"}
milight/0xEF3/cct/3 {"state":"OFF"}
milight/states/0xEF3/cct/3 {"state":"OFF","brightness":255,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/0xEF3/cct/3 {"state":"ON"}
milight/0xEF3/cct/3 {"state":"OFF"}
milight/states/0xEF3/cct/3 {"state":"OFF","brightness":255,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/0xEF3/cct/3 {"state":"ON"}`

sidoh, 2020-06-28:
Your `mqtt_update_topic_pattern` and `mqtt_topic_pattern` are set to the same thing. This is going to cause a feedback loop, so you'll need to make them different. See [the README](https://github.com/sidoh/esp8266_milight_hub#more-detail-on-mqtt_topic_pattern) for more details on MQTT settings.

akawoj, 2020-06-28:
> Your `mqtt_update_topic_pattern` and `mqtt_topic_pattern` are set to the same thing. This is going to cause a feedback loop, so you'll need to make them different. See [the README](https://github.com/sidoh/esp8266_milight_hub#more-detail-on-mqtt_topic_pattern) for more details on MQTT settings.

You were right. Thank you and sorry for the obvious mistake :)

sidoh, 2020-06-28:
No worries, MQTT setup is a little convoluted. Glad you got it working.

@daniel9522, I think your issue is the same, so going to close. Please bump the ticket if you're still having issues and we can take another look.

closed 2020-06-28 by sidoh