sidoh/esp8266_milight_hub #641: Option to have configurable JSON file for MQTT

markbooker, 2020-06-18:
No sure how feasible or achievable this is, but as the wall plates are so nice, I was looking to use these to control Tasmota flashed lights.

Whilst I can do a little bit of coding, I think this may be beyond my skillset.

Would it be possible to add a feature to select a remote / wallplate function to say it is sending to a tasmota device and format the mqtt message accordingly?

I know it is possible with another piece of software running on another device to take the mqqt responses and send additional messages to these devices but I thought it would be cleaner to have it in one place.

Tasmota MQTT needs to see:
POWER: ON/OFF
Dimmer: 0-100
Color: 000000-FFFFF

Thanks for your consideration

sidoh, 2020-06-18:
Hi @markbooker,

The primitives are there to support this sort of thing (see MQTT fields in settings). Hex color is already supported, for what it's worth.

I won't have time to look at this for a while, but here's a reference commit adding support for a field. The others you mentioned should be pretty straightforward to pattern mach off of from this:

https://github.com/sidoh/esp8266_milight_hub/commit/6c16e258e597c4e7503e0fe886683cda04879b67

markbooker, 2020-06-18:
@sidoh,

Thanks for the consideration and response, I did already investigate the additional fields available and yes the combined mqtt json response does send all the values needed, sadly the limitation on tasmota does not allow me to interpret the full json string, hence the request to be able to send the right fields as an option from this.

I would be great if you do get time at some point to look into it.

I have tested each of the fields individually and confirmed they work with the tasmota firmware.

Thanks again for such a great project this truly helps to keep our home automation in the home where it belongs.

for reference, I have separately tested the following fields with tasmota firmware:
color mapped to color : works as expected
status or state mapped to power : works as expected
level mapped to dimmer : works as expected

I will also test the scenes and update once I have

servim, 2020-06-25:
@markbooker, My English is very poor, I did not understand what you want to do, but I'm very interested :)
Please write in more detail what and why you want to configure?
Thank!