sidoh/esp8266_milight_hub #642: Collaboration with ESPHome

randyarrasmith, 2020-06-23:
It would be awesome if I could see my milight hub esp8266 in esphome.  I suspect many of us using this are familiar with esphome and home assistant.  In my opinion it would be a cool collaboration to be able to manage the hub and perhaps configure groups, control lights, etc using esphome.   

just a thought :)   love the work you have done!

sidoh, 2020-06-25:
Hi @randyarrasmith,

Thanks for the suggestion.

I have pretty limited understanding of ESPHome, so hard for me to feel confident I'm onto the right thing when I'm imagining how ESPHome and espMH could integrate. Could you perhaps go into more detail?

randyarrasmith, 2020-06-25:
@sidoh   Hi Chris.  ESPHome is a user friendly interface to manage ESP8266/32 devices all in one place and using simple yaml configuration files with a built in API to home assistant.  Easy to make changes and update devices OTA (over the air), etc.  It is very neat.   It is built on platformIO and so is your milight hub.  I realize at a glance this may not be a perfect match since the hub is not a sensor but I am thinking something like yaml config for the different light groups, OTA for updates, etc.  Defining the lights in ESPHome would make them automatically available in HA, then do something like "milight:" keyword to have it include the needed bridge component(s) then configuration of the bridge ip.  I hope I am explaining this ok.  It seems like an exciting idea (to me) and perhaps you can collaborate with Otto Winters (ESPHome).  
https://github.com/esphome/esphome
https://esphome.io

what made me think of this is that I often ticker with esp(s) via ESPHome and forget I have this other one (the milight hub) sitting out there.  I set it up like 2 years ago and I was recently demoing some things to a friend and one of the cool things is to change my front yard up-lights different colors, especially for holidays.  So then I thought I should check if there is an update to the hub and of course there is.  Man would it be cool if that showed in esphome with my other nodes and show an update was available.  Then it occurred to me I could define the lights there also and not in HA directly.  And then I could do an automation/transition from the esp also and not from HA.  Who knows what else I am not even thinking of right now.  :)     

just an idea....   I appreciate the work you have done.  And love the use of the hub in my system

sidoh, 2020-06-30:
Gotcha, that makes sense. I have used ESPHome before, but was wondering if there was some mode of operation with it I was unfamiliar, but sounds like not.

Integrating with it in this way would mean pulling the core milight stuff out of this project and adapting it to work with ESPHome's APIs, which would be a pretty huge undertaking. ESPHome is great for simple stuff, but I've found that I hit its limitations pretty quickly as what I'm trying to do becomes more complex. I'm therefore inclined to keep espMH its own thing.

Certainly supportive of someone else who wants to take this on, but this isn't something I see myself doing. Going to close, but feel free to keep the discussion going :)

closed 2020-06-30 by sidoh

hencou, 2021-10-15:
Hi,

I have been using espMH for 4 years now, and am very happy with it!.

I have 20 (slightly modified) espMH devices in use: behind every wall switch, on that way I can control the lamps by the wallswitch, and via the Homeassistant (MQTT).

Also use ESPhome for a few other projects, and the easy management of those, and OTA also really appeals to me. I have therefore made an attempt at what Sidoh says: using the core from espMH in ESPHome. A first draft has been successful: see here: 
https://github.com/hencou/esphome_milight_wallswitch
Maybe it helps somebody who is searching for an example like this.

shbatm, 2022-01-01:
@hencou @sidoh Just want to say thank you for both of your work on this project. I was able to get @hencou's version up and running in ESPHome on an ESP32 D1 Mini and can confirm it's working well so far controlling lights from Home Assistant. Using ESPHome was much easier to get it up on an ESP32 than continuing the efforts started on #725. 

@hencou - Let me know if there's anything else you'd like me to test. Adding remote receiving support would be nice too.

hencou, 2022-01-01:
> @hencou - Let me know if there's anything else you'd like me to test. Adding remote receiving support would be nice too.
Receiving support should work, in the "esphome_components" version. When you compile ESP Home with loggin level "DEBUG" then you will see the received commands in the logging console.

Receiving support should work, in the "esphome_components" version. When you compile ESP Home with logging level "DEBUG" then you will see the received commands in the logging console. The state of the light in homeassistant will also update on the received commands.

shbatm, 2022-01-02:
> Receiving support should work, in the "esphome_components" version.

Awesome, thanks. I didn't look that deep into that side yet. I'll play with it more, I'd like to eventually be able to receive commands from a separate remote and send them as events to Home Assistant (to control non Mihome lights). If the support is in there I should be able to do it with templates when I get to it. Thanks again.

poudenes, 2022-12-03:
Hi ALl,

is this still in process and or is there already running ESP home MH version out there?
If so. Is there a way to move the correct config and paired bulbs into the ESP home MH ?

poudenes, 2022-12-03:
> Yes, it has been running satisfactorily here for almost a year now. No need to re-pair the bulbs, just get the same DeviceIDs, GroupIDs and BulbType in your yaml config. See the yaml example: https://github.com/hencou/esphome_components/blob/main/example_milight.yaml

Nice, can this also used on a Wemos D1 for example and add all lights in the config?
I have 34 lights now running on the Mlight Hub of Sidoh.

shbatm, 2022-12-04:
Yep. I have 2 bridges running ~20 lights on Esp32s with ESPHome. 

The biggest disadvantage has been that you have to preplan the groups and add them to the config and reflash, but definitely worth the seamless integration with HA /Esphome dashboard. And with the new Text entity coming in HA, I'm sure there's going to be a way to replicate the Web UI from this repo.

poudenes, 2022-12-04:
> Yep. I have 2 bridges running ~20 lights on Esp32s with ESPHome.
>
> The biggest disadvantage has been that you have to preplan the groups and add them to the config and reflash, but definitely worth [...]

Nice,

Starting with a second Wemos D1 R2 module to create the hub. Get lots of issues. Wifi connection issues.
Lost API issues etc... OTA uploads stuck on 8,15,67 or 90%

Have 8 ESP modules (wall switches) they work without any problem... Even far away from the router.
Wemos D1 is even 1 meter from router and have issues

```
#########################################################
# Below all fixed settings for Milight PCB
#########################################################
substitutions:
  devicename: milight-hub-test
  friendly: Milight Test
  ip: 192.168.100.210
#########################################################

esphome:
  name: ${devicename}
  libraries:
    - milight=https://github.com/hencou/esphome-milight

external_components:
  - source: github://hencou/esphome_components
    components: mi

esp8266:
  board: d1_mini
  restore_from_flash: true

api:
  encryption:
    key: !secret api_key

time:
  - platform: homeassistant
    id: homeassistant_time

preferences:
  flash_write_interval: 1min
  
logger:
  
ota:
  safe_mode: true
  password: !secret password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  reboot_timeout: 0s
  manual_ip:
    static_ip: ${ip}
    gateway: 192.168.100.1
    subnet: 255.255.255.0
    dns1: 192.168.100.1
  ap:
    ssid: ${devicename}
    password: !secret password
    channel: 4

#########################################################
# Below specific for Milight settings
#########################################################
mi:
[... 341 lines cut ...]

  - platform: mi
    id: toilet2
    name: "Toilet 2 test"
    device_id: 0xAAAE
    group_id: 2
    remote_type: rgb_cct
    default_transition_length: 0s

  - platform: mi
    id: toilet3
    name: "Toilet 3 test"
    device_id: 0xAAAE
    group_id: 3
    remote_type: rgb_cct
    default_transition_length: 0s

  - platform: mi
    id: toilet_all
    name: "Toilet all test"
    device_id: 0xAAAE
    group_id: 0
    remote_type: rgb_cct
    default_transition_length: 0s
```

Linkenelis, 2022-12-04:
I use iobroker, it has a milight adaptor. As I am running 3 device ID's on esp8266, 3 instances of the adaptor are used.
[image]
It looks like this:
[image]
gestures up or down for brightness, click on/off (that's enough for the family ;-))

hencou, 2022-12-04:
> Nice,
>
> Starting with a second Wemos D1 R2 module to create the hub. Get lots of issues. Wifi connection issues. Lost API issues etc... OTA uploads stuck on 8,15,67 or 90%
>
> Have 8 ESP modules (wall [...]

Strange, what happens when you try with a few light devices configured in yaml? Maybe the Wemos D1 devices are less powerfull then ESP32s. I am also using Wemos D1 modules, with a max of 5 devices (never tried more) and that's working stable all the times.

poudenes, 2022-12-04:
I'm using wemos d1 with original milight image and all light are working. Maybe esphome use lot more power? I will try tomorrow with less lights and see what happen. Then add more to.it u till it become unstable.

poudenes, 2022-12-08:
This gives me a good result. Sometimes 1 or 2 single bulbs not go off but after 2 to 3 seconds they react.
```
  listen_repeats: 1
  state_flush_interval: 0
  packet_repeats: 25
  packet_repeats_per_loop: 25
  packet_repeat_throttle_threshold: 0
  packet_repeat_throttle_sensitivity: 10
  packet_repeat_minimum: 12
```
Maybe someone can tell me what function every setting is. I can't any logical explanation to understand what each setting do. So I change and change and update the ESP32 a 100 times.. 
Because my last post was also a great result but with totally different settings:

```
  listen_repeats: 2
  state_flush_interval: 1500
  packet_repeats: 50
  packet_repeats_per_loop: 75
  packet_repeat_throttle_threshold: 800
  packet_repeat_throttle_sensitivity: 10
  packet_repeat_minimum: 25
```