sidoh/esp8266_milight_hub #644: unpairing when pairing

s1nbad, 2020-07-01:
Hoping you can help

I have a room with 30 gu10 milight bulbs all on one electrical circuit
I pair them all to a virtual device_id
3 of the bulbs have become unpaired 
When I pair the 3 bulbs the other 27 unpair
When I pair the 27 bulbs the 3 unpair again

I don't have the same issue using the remote fut089 or the wall unit B8 the 3 bulbs pair and the other 27 remain paired - this behavior only occurs using the bridge.

any ideas ?

sidoh, 2020-07-01:
Oh man, that's super obnoxious. Sorry to hear about that.

Are you able to include some packet lots so we can spot any differences between your remote and espMH? 

For what it's worth -- I use FUT089 with FUT015 bulbs, and I'm pretty sure they don't have this behavior.

s1nbad, 2020-07-05:
Having a look at the results of sniffing it appears that the bridge is sending out a sequence of 5 x 9byte packets for both pairing and unpairing.  Whilst for the remote, which is manual, 3 x 9byte packets for pairing and 6 x 9byte packets for unpairing.

sidoh, 2020-07-08:
The number of packets shouldn't matter, but the structure might. Can you please paste the packet logs? :)

sidoh, 2020-11-01:
Going to close this since it's been open for so long, please feel free to comment to reopen.

closed 2020-11-01 by sidoh