sidoh/esp8266_milight_hub #645: Problems with remote fowarding and effect speed

camcs1, 2020-07-06:
### Describe the bug

If you are using remote forwarding and hold speed up or down on a physical remote it puts the lights into nightmode after a few seconds 

### Steps to reproduce

Use the remote forwarding example for Home Assistant, then hold speed down 

### Expected behavior

Behaves as expected in that it slows the effect speed, however the switch to nightmode is unwanted 

### Setup information
I am using a NodeMCUv3 Clone with a  NRF24L01+PA+LNA SMA with external antenna. 

For MiLight hardware I have: a FUT092 & B4 as remotes and FUT012, FUT039, FUT037 & FUT014 as lights. Both remotes trigger the bug and all lights also suffer from it. 

#### Firmware version
1.10.5 & 1.10.6

#### Output of http://milight-hub.local/about

```json
{"firmware":"milight-hub","version":"1.10.6","ip_address":"192.168.0.21","reset_reason":"External System","variant":"esp12","free_heap":15944,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```json
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":1440,"mqtt_server":"192.168.0.22:1883","mqtt_username":"cam","mqtt_password":"","mqtt_topic_pattern":"milight/commands/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"/milight/state/:device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"Fast blip","led_mode_operating":"Off","led_mode_packet":"Off","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"192.168.0.21","wifi_static_ip_gateway":"192.168.0.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[6773,7029,7285],"gateway_configs":[],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{"Front":["rgb_cct",7029,4],"Lamp":["rgb_cct",7029,2],"Top":["rgb_cct",7029,3],"Unit":["rgb_cct",7029,1]}}
```

###Home Assistant Setup 

```
- alias: MiLight Forwarder
  initial_state: true
  trigger:
    platform: mqtt
    topic: milight/updates/0x1A75/rgb_cct/+
  action:
  - service: mqtt.publish
    data_template:
      topic: milight/commands/0x1B75/rgb_cct/{{ trigger.topic.split('/')[4] }}
      payload_template: '{{ trigger.payload }}

        '
  id: 5326fdd5087a4adcb7758366a55fcbe5
- alias: MilightAll
  initial_state: true
  trigger:
    platform: mqtt
    topic: milight/updates/0x1A75/rgb_cct/0
  action:
  - service: mqtt.publish
    data_template:
      topic: milight/commands/0x1B75/rgb_cct/1
      payload_template: '{{ trigger.payload }}

        '
  - service: mqtt.publish
    data_template:
      topic: milight/commands/0x1B75/rgb_cct/2
      payload_template: '{{ trigger.payload }}

        '
  - service: mqtt.publish
    data_template:
      topic: milight/commands/0x1B75/rgb_cct/3
      payload_template: '{{ trigger.payload }}

        '
  - service: mqtt.publish
    data_template:
      topic: milight/commands/0x1B75/rgb_cct/4
      payload_template: '{{ trigger.payload }}

        '
  id: bfba1df920894ffaa8525d3147c854ed
- alias: MiLight Wall Remote
  initial_state: true
  trigger:
    platform: mqtt
    topic: milight/updates/0x6B5F/rgb_cct/+
  action:
  - service: mqtt.publish
    data_template:
      topic: milight/commands/0x1B75/rgb_cct/{{ trigger.topic.split('/')[4] }}
      payload_template: '{{ trigger.payload }}

        '
  id: e51a5125247145c697190b3470a4e586
- alias: MilightWallAll
  initial_state: true
  trigger:
    platform: mqtt
    topic: milight/updates/0x6B5F/rgb_cct/0
  action:
  - service: mqtt.publish
    data_template:
      topic: milight/commands/0x1B75/rgb_cct/1
      payload_template: '{{ trigger.payload }}

        '
  - service: mqtt.publish
    data_template:
      topic: milight/commands/0x1B75/rgb_cct/2
      payload_template: '{{ trigger.payload }}

        '
  - service: mqtt.publish
    data_template:
      topic: milight/commands/0x1B75/rgb_cct/3
      payload_template: '{{ trigger.payload }}

        '
  - service: mqtt.publish
    data_template:
      topic: milight/commands/0x1B75/rgb_cct/4
      payload_template: '{{ trigger.payload }}

        '
  id: 4eb35eae47854330b973a335d33409e5
```
### Additional context
You might of noticed I'm using the ESP12 binary, did this as a troubleshooting step however the bug was also present in the NodeMCU binary.

sidoh, 2020-07-13:
This is probably not something I'll have the chance to look into for a while. Here's the relevant code, for anyone brave enough to go spelunking :)

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/RgbCctPacketFormatter.cpp#L110

sidoh, 2020-07-20:
Appreciate you taking a stab at it, @camcs1!

If it helps, there's some very light documentation on the protocol here: https://blog.christophermullins.com/2017/03/18/reverse-engineering-the-new-milightlimitlessled-2-4-ghz-protocol/

Let me know if there's anything I can help with :)