sidoh/esp8266_milight_hub #646: Switching off with remote after switching on with HA - is it possible to have proper state in HA?

akawoj, 2020-07-11:
Hello!

Here is the scenario:
1. The LED are switched on with migliht_hub via Home Assistant interface
2. The LED are switch off with milight standard radio-remote
3. Status of the LEDs in HA interface is still shown as "on". **Is there any way for milight_hub** to pass info to HA, that the LED where switched off? Am I missing some configuration?

sidoh, 2020-07-13:
Seems like this is what you're talking about?

https://github.com/sidoh/esp8266_milight_hub/wiki/Using-Milight-Remote-with-HomeAssistant

akawoj, 2020-07-14:
Yes!

Sorry, I've missed this documentation section somehow...

closed 2020-07-14 by akawoj

sidoh, 2020-07-14:
No worries, the docs aren't exactly perfectly organized :)