sidoh/esp8266_milight_hub #648: Log Error in Home Assistant Log

andreanilorenzo, 2020-07-19:
### Describe the bug

I Have this in Home Assistant Log

`Unsupported effect value received
0:33:14 – MQTT (WARNING) - il messaggio si è verificato per la prima volta alle 0:30:17 e compare 5 volte
Invalid color value received
0:33:14 – MQTT (WARNING) - il messaggio si è verificato per la prima volta alle 0:30:17 e compare 4 volte
Invalid color temperature value received
0:33:14 – MQTT (WARNING) - il messaggio si è verificato per la prima volta alle 0:30:17 e compare 4 volte
Error parsing value: 'dict object' has no attribute 'color_temp' (value: {"state":"OFF","status":"OFF","brightness":255,"level":100,"hue":38,"saturation":100,"color":"#FFA100","bulb_mode":"color"}, template: {{ ((value_json.color_temp)) | int }})
0:33:14 – helpers/template.py (ERROR) - il messaggio si è verificato per la prima volta alle 0:30:17 e compare 4 volte
Invalid brightness value received
0:30:17 – MQTT (WARNING)`

#### Firmware version
1.10.6 (nodemcuv2)

#### Output of http://milight-hub.local/about

`{"firmware":"milight-hub","version":"1.10.6","ip_address":"192.168.1.53","reset_reason":"Software/System restart","variant":"nodemcuv2","free_heap":16880,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}`

#### Output of http://milight-hub.local/settings

`{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.1.60:1883","mqtt_username":"mqtt_user","mqtt_password":"123456","mqtt_topic_pattern":"milight/set/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status/lorenzo","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":400,"mqtt_debounce_delay":400,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub-lorenzo","rf24_power_level":"MAX","rf24_listen_channel":"MID","wifi_static_ip":"192.168.1.53","wifi_static_ip_gateway":"192.168.1.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":900,"rf24_channels":["MID"],"device_ids":[16,17,18],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type","oh_color","hex_color"],"group_id_aliases":{"Milight-Bagno":["rgb_cct",16,1],"Milight-Cucina":["rgb_cct",17,1],"Milight-Soggiorno":["rgb_cct",18,1]}}`

sidoh, 2020-07-20:
try adjusting the white temperature, either in HASS or in the espMH UI. it's not transmitting `color_temp`, probably because a value isn't set yet.

andreanilorenzo, 2020-10-29:
I haven't been able to solve the problem yet, I have the log full of this error and the milight lights have not been responding well to commands for some time.
Which mqtt template should I use for the rgb-cct bulb?

sidoh, 2020-11-01:
The template looks fine. Have you tried updating the color temperature for a bulb?

andreanilorenzo, 2020-11-01:
I have a "E27 RGB+CCT 6W MiLight" lamp bulb

Today I did some testing, the problem is here:
- in Home Assistant configuration.yaml I use MQTT Template Light (which does not support HS color)
https://www.home-assistant.io/integrations/light.mqtt/
- so in the template I can only insert RGB colors, but the Hub returns only HSV colors
- to eliminate errors in the log I entered a value of 255 in the RGB color template, but this obviously does not update the correct color of the lamp on Home Assistant
- why does the Hub not return RGB colors in JSON in MQTT?

andreanilorenzo, 2020-11-01:
This is the template:

`light:
  - platform: mqtt
    name: "Color Ingresso"
    schema: template
    command_topic: "milight/set/0x12/rgb_cct/1"
    state_topic: "milight/states/0x12/rgb_cct/1"
    unique_id: 56a136d1-2801-4c5o-9b2a-6a0f5b0d09a3     # https://www.uuidgenerator.net/version4
    availability_topic: "milight/client_status/lorenzo"
    <<: &MILIGHT_PARAMS
      qos: 1
      optimistic: false
      retain: false
      payload_available: "connected"
      payload_not_available: "disconnected"
      effect_list:
        - night_mode
        - white_mode
        - color
        - 0
        - 1
        - 2
        - 3
        - 4
        - 5
        - 6
        - 7
        - 8
      command_on_template: >
        {"state":"ON"
        {%- if brightness is defined -%}
        , "brightness": {{ brightness }}
        {%- endif -%}
        {%- if transition is defined -%}
        , "transition": {{ transition }}
        {%- endif -%}
        {%- if color_temp is defined -%}
        , "color_temp": {{ color_temp }}
        {%- else -%}
            {%- if red is defined and green is defined and blue is defined -%}
            , "color": {"r":{{ red }},"g":{{ green }},"b":{{ blue }} }
            {%- endif -%}
        {%- endif -%}
        {%- if effect is defined -%}
          {%- if effect == "night_mode" -%}
          , "effect": "night_mode"
          {%- elif effect == "color" -%}
          , "effect": "0"
          {%- else -%}
          , "effect": "{{ effect }}"
          {%- endif -%}
        {%- endif -%}
        }
      command_off_template: >
        {"state":"OFF"
        {%- if transition is defined -%}
        , "transition": {{ transition }}
        {%- endif -%}
        }
      state_template: '{{ value_json.state | lower }}'
      brightness_template: >
        {% if ('brightness' in value_json) and (value_json.brightness != Null) %}
          {{ value_json.brightness }}
        {% endif %}
      color_temp_template: >
        {% if ('color_temp' in value_json) and (value_json.color_temp != Null) and (value_json.color_temp != None) %}
          {{ ((value_json.color_temp)) | int }}
        {% else %}
          260
        {% endif %}
      red_template: >
        {% if ('color' in value_json) and (value_json.color != Null) and (value_json.color != None) and (value_json.color["r"] != Null) %}
          {{ value_json.color["r"] | int }}
        {% else %}
           {{ 255 | int }}
        {% endif %}
      green_template: >
        {% if ('color' in value_json) and (value_json.color != Null) and (value_json.color != None) and (value_json.color["g"] != Null) %}
          {{ value_json.color["g"] | int }}
        {% else %}
          {{ 255 | int }}
        {% endif %}
      blue_template: >
        {% if ('color' in value_json) and (value_json.color != Null) and (value_json.color != None) and (value_json.color["b"] != Null) %}
          {{ value_json.color["b"] | int }}
        {% else %}
          {{ 255 | int }}
        {% endif %}
      effect_template: >
        {% if ('bulb_mode' in value_json) and (value_json.bulb_mode == "color") %}
          {{ value_json.bulb_mode }}
        {% elif ('effect' in value_json) and (value_json.effect != Null) and (value_json.effect != None) %}
          {{ value_json.effect }}
        {% else %}
          'color'
        {% endif %}
      min_mireds: 153
      max_mireds: 370
`

sidoh, 2020-11-01:
It does return RGB colors. The relevant fields are `color`, `computed_color` (this is the one you should use for HomeAssistant), or `hex_color`.

The default values are selected to work with HomeAssistant. Try this:

```
"group_state_fields":["state","brightness","level","color_temp","bulb_mode","computed_color","effect"]
```

See the documentation for more details:

https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant#manually-hass--084

(there's an example config there -- it should not need to be nearly as complex as what you've got in your last post).

andreanilorenzo, 2020-11-01:
Yes, I read all the documentation and set the template like this, but I had errors in the log and so I changed.
I do not use Autodiscovery

This is what the Hub writes to the MQTT server:

`{
  "state": "ON",
  "brightness": 255,
  "level": 100,
  "hue": 244,
  "saturation": 100,
  "color": "#1000FF",
  "bulb_mode": "color"
}`

`{
  "state": "ON",
  "brightness": 255,
  "level": 100,
  "hue": 240,
  "saturation": 100,
  "color": null,
  "bulb_mode": "color"
}`

`{
  "state": "ON",
  "brightness": 255,
  "level": 100,
  "kelvin": 96,
  "color_temp": 361,
  "bulb_mode": "white",
  "color": {
    "r": 255,
    "g": 255,
    "b": 255
  },
  "effect": "white_mode"
}`

sidoh, 2020-11-01:
I use RGB+CCT bulbs with HomeAssistant myself using those instructions, and have no issues. The three messages you pasted all look pretty different -- why is that? Are those retained messages? Try setting `group_state_fields` to exactly what I pasted in my last message.

andreanilorenzo, 2020-11-01:
Ok, I checked now, I had left other fields active in "group_state_fields" and the "color" value was not written in MQTT.

I changed the field by selecting only the values you indicated and it works (obviously as you told me):
"group_state_fields": ["state", "brightness", "level", "color_temp", "bulb_mode", "computed_color", "effect"]

sidoh, 2020-11-03:
Excellent!

closed 2020-11-03 by sidoh