sidoh/esp8266_milight_hub #649: Report status changes on MQTT as well

markg85, 2020-07-31:
Hi,

Currently there is `mqtt_state_topic_pattern` but it's purpose is to send a full new state for the light to go to.

I would like to request a topic that "tells" me the new state of a light bulb after it has been set.
In my case it is a simple node.js application that currently does the state changes. I "can" change it to do the broadcasting there myself, but i really would prefer it to be build in as it seems like the perfect place to be imho. And would be useful for others too.
It would allow other applications using this library to fairly accurately know the light state. Yes, i'm aware the state change would be "best effort" based as the milight protocol is one-way only...

I propose a topic name of:
`mqtt_state_feedback_topic_pattern`

Where it should be set to something like this:
`milight/statefeedback/:hex_device_id/:device_type/:group_id`

The full current new state should be put on that channel.

What's your thought on this?
Also, I can make a patch for this if that would help. If you'd accept such a feature that is :)

Cheers,
Mark

markg85, 2020-07-31:
Nope. Forget this one. It's there already :)
Closed. Sorry for the noise.

closed 2020-07-31 by markg85

sidoh, 2020-07-31:
Glad you found it!

The MQTT docs in the README are pretty muddled, so I take the blame for the confusion. :)