sidoh/esp8266_milight_hub #650: Query about 'level' scaling and possibility to have a direct 1-1 scale between 0-100

DigiH, 2020-08-15:
First of all, thanks a lot for a brilliant replacement for my several Milight bridges (v3 and v6) with the possibility of having an NRF24L01+ with external antenna attached, and all the configuration options.

I have set it all up in the last two days and am currently playing around with using it in openHAB directly with MQTT and with  Skinah's EspMilightHub binding.

One thing that came to my attention though is the slight scaling of the 'level' channel, which doesn't allow a true setting of 0, and therefor also scales the higher values. 

I assume this is because of the possibility of recognising remotes, which might not allow a level of 0. 

But whenever the dimmer slider in openHAB is being set to, let's say, 50, and it then jumps to 52 it is sort of irritating. Also precise values in the very low range are not really possible, which I like to use when watching films or late at night, i.e. sending command {"level":2} and getting back state …,"level":4,…

Is there already a possibility, which I couldn't find, to have a direct 1-1 assignment to 'level', or would it be possibly to include a setting which allows this, when there are not remotes in existence which might need to be considered?

Many thanks

sidoh, 2020-08-17:
Hi DigiH, thanks for the question.

* You _should_ be able to set a value of 0. What happens when you do?
* Level _should_ be 1-1. For example, check out these MQTT logs --

```
milight/0x2666/rgb_cct/1 {"level":3}
milight/states/0x2666/rgb_cct/1 {"brightness":8,"level":3,"color":{"r":255,"g":255,"b":255}}
milight/0x2666/rgb_cct/1 {"level":4}
milight/states/0x2666/rgb_cct/1 {"brightness":10,"level":4,"color":{"r":255,"g":255,"b":255}}
milight/0x2666/rgb_cct/1 {"level":0}
milight/states/0x2666/rgb_cct/1 {"brightness":0,"level":0,"color":{"r":255,"g":255,"b":255}}
```

Brightness is _not_ 1-1 because the state is down-converted to a 0-100 range.

I guess in other words, I'm not seeing this behavior. Could you maybe provide some MQTT/REST commands I could use to reproduce?

DigiH, 2020-08-17:
Hi Sidoh,

Thanks for the reply. My main level jumping issue was with the RGBW bulbs not the RGB+CCT ones. Still reproducible here for me, 25 to 24, 2 to 4, 15 to 16 ...

milight/commands/0x3249/rgbw/2  {"level":50}
milight/states/0x3249/rgbw/2  {"state":"ON","level":52,"hue":21,"bulb_mode":"color"}

milight/commands/0x3249/rgbw/2  {"level":25}
milight/states/0x3249/rgbw/2  {"state":"ON","level":24,"hue":21,"bulb_mode":"color"}

milight/commands/0x3249/rgbw/2  {"level":2}
milight/states/0x3249/rgbw/2  {"state":"ON","level":4,"hue":4,"bulb_mode":"color"}

milight/commands/0x3249/rgbw/2  {"level":15}
milight/states/0x3249/rgbw/2  {"state":"ON","level":4,"hue":16,"bulb_mode":"color"}

Always with the same up/down jumps. Haven't done a whole 0-100 test sequence yet though ;)

I still have some RGBW bulbs in places where I don't really need any colour saturation control, but if this is a non-fixable issue I might just have to swap them all over to RGB+CCT.

> For RGB+CCT a level of 0 seems impossible.

What I meant here is that when setting the level to 0, the bulbs (RGBW and RGB+CCT ) still have a very low dimmed light. Is this a failsafe implementation in the code, to avoid a true level 0 when it gets a "state":"ON" command, or is it a limitation of the bulbs themselves, which still have a dimm light, even when they get a level 0 command?

Maybe best to explain where I'm coming from - whenever I send an OFF command to the bulbs, I dimm down the level to 0 first. I do this to avoid a brief bright flash when the bulb was on a high level, e.g. 100, before being turned off, and then gets a low level command when turned on the next time. Especially when the lights get turned on automatically when I get up at night.

With level 0 not being a true level 0, i.e. no light at all, I have already implemented some failsafe myself in code where I do the level = 0, then state = OFF manually, but having the items also in a groups in openHAB, whenever the groups get an OFF command, there is always a short dimm flash of the bulbs.

Also when the lights do come back on, they always do a short wobble, from OFF to level 1, then to the level which was sent. Whereas I would have expected a direct OFF (level 0) to sent the sent level.

I hope this explains the situation a bit better.

sidoh, 2020-08-17:
Ah, I'm following now. Thanks for the clarifications.

#### Re: Funky brightness scaling

The RGBW protocol uses a very strange brightness scale -- the full reasoning for this is inscrutable, it's essentially because they tried to cram the [brightness value into 5 bits](https://github.com/sidoh/esp8266_milight_hub/wiki/RGBW). 

It's not ideal that users are exposed to this, but it's a consequence of an intentional architectural decision. Rather than state being kept based on commands received, it's based on raw milight packets that are _sent_. I can explain why this is the case if you're curious, but I'll omit it by default for the sake of brevity.

So the flow here is basically:

1. You send the `{"level":25}` command
2. espMH translates this into an RGBW packet sent to the bulbs, mapping 25 on a [0,100] scale to 6 on a [0,25] scale
3. That packet is passed through the state handling system, which recognizes that the brightness has been set to 6 on a [0,25] scale, which corresponds to 24 on a [0,100] scale
4. The updated state is transmitted via MQTT showing `{"level":24}`

#### Re: level 0

With all milight bulbs I'm aware of, brightness=0 is the dimmest possible setting (outside of "night mode"), but is still on.

The command handler is smart enough to interpret the command `{"state":"off","level":0}` in the order that you'd hope -- it first dims the bulb to level=0, and then turns it off. Similarly, `{'state":"on","level":100}` will turn the bulb on (it'll be in whatever brightness state it was before it was last turned off), and then ramp the brightness up to 100.

DigiH, 2020-08-17:
> Rather than state being kept based on commands received, it's based on raw milight packets that are sent

Thanks, that clears up this issue and reason behind it for me.

> The command handler is smart enough to interpret the command {"state":"off","level":0} in the order that you'd hope -- it first dims the bulb to level=0, and then turns it off. Similarly, [...]

Great, now I just have to adjust some of my routines in openHAB to accommodate for the slightly new behaviour, also with it all working together with  Skinah's EspMilightHub binding, needing proxy switches for the Switch items with :level channels now, which are assigned to groups. 

Thanks again for all the clarifications. This got me on the way to have everything working exactly how I want it again :)

Closing as 'query all cleared up'

closed 2020-08-17 by DigiH