sidoh/esp8266_milight_hub #653: Support for MIBOXER FUTD02

jorismathijssen, 2020-09-02:
### What is the model number of the device you're trying to control?
FUTD02 DMX512

### What firmware version(s) have you tried? 
None

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 
None
### Which radio type are you using? (RGBW, RGB+CCT, etc.) 
RGB

### Have you tried controlling the device with a physical remote?
Yes, Using the FUT089/B8

I have a quick question, im currently working with a FUTD02 (DMX512) receiver. I have the lights working with the remote (FUT089/B8) and with the app using the MIBOXER WL-Box1. 

Is it possible to use this milight ESP8266 hub to control my lights? Are there other options for me to control my lights? Is the MIBOXER WL-Box1 only usable through the Tuya cloud API?

I hope someone can help me with my questions, thank you in advance!

camcs1, 2020-09-14:
The way I understand it is that this program supports remotes rather than bulbs/controllers.

 So if you can use a FUT089/B8 to control them then this program should work for you (as it's just emulating the physical remotes)

sidoh, 2020-11-01:
Sorry for the late reply.

@camcs1 is correct. 

The remote in the photo looks like FUT092 (called "RGB+CCT" in this project). So yes, should be possible.

closed 2020-11-01 by sidoh