sidoh/esp8266_milight_hub #654: White not showing back to Hassio

livewild1988, 2020-09-12:
hello,
when changing colour of the device / controller it works as requested but as soon as you enable white the controller reacts but hassio still shows the last colour used on the state icon. so the lights could be white but the symbol would show red green or blue.

also is there anyway of using rgb as rgbw 255,255,255,255 ive switched from dmx to milight but am unable to make walm white daylight now using this white channel.

thanks chris

sidoh, 2020-10-09:
If I'm following, I think `computed_color` in `mqtt_group_state` fields does what you want.

sidoh, 2020-10-15:
@livewild1988 -- can you let me know if the above is what you're looking for?

livewild1988, 2020-10-15:
Hello that has worked for me but I’m unable to have rgbw values all on at
the same time. It’s either RGB or W is there a way round this

Thank you for responding to me

sidoh, 2020-10-15:
Great! Gonna close this out then.

If I'm understanding the question correctly, all of the Milight devices I'm familiar with only support having one of (A) RGB LEDs or (B) WW LEDs on at one time.

closed 2020-10-15 by sidoh