sidoh/esp8266_milight_hub #656: Uncompilable with PIO 5 (both firmware and website)

maxgerhardt, 2020-10-03:
### Describe the bug

When downloading the repository and attemping to build an environment, e.g. for `nodemcuv2`, the firmware build as well as the website build are broken.

### Steps to reproduce

```
> git clone --recursive https://github.com/sidoh/esp8266_milight_hub.git
> cd esp8266_milight_hub
> pio run -e nodemcuv2

Processing nodemcuv2 (platform: espressif8266@~1.8; framework: arduino; board: nodemcuv2)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Attempting to build webpage...
npm WARN deprecated gulp-util@3.0.8: gulp-util is deprecated - replace it, following the guidelines at https://medium.com/gulpjs/gulp-util-ca3b1f9f9ac5
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated minimatch@2.0.10: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated natives@1.1.6: This module relies on Node.js's internals and will break at some point. Do not use it, and update to graceful-fs@4.x.
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated minimatch@0.2.14: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated graceful-fs@1.2.3: please upgrade to graceful-fs 4 for compatibility with current and future versions of Node.js
npm WARN deprecated resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated
npm WARN deprecated urix@0.1.0: Please see https://github.com/lydell/urix#deprecated
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN esp8266-milight-hub-web@1.0.0 No description
npm WARN esp8266-milight-hub-web@1.0.0 No repository field.

b'added 426 packages from 261 contributors and audited 426 packages in 10.582s\n\r\n4 packages are looking for funding\r\n  run `npm fund` for details\n\nfound 9 vulnerabilities (3 low, 6 high)\n  run `npm audit fix` to fix them, or `npm audit` for details\n'
fs.js:35
} = primordials;
    ^

ReferenceError: primordials is not defined
    at fs.js:35:5
    at req_ (C:\Users\Maxi\Desktop\repo\esp8266_milight_hub\web\node_modules\natives\index.js:143:24)
    at Object.req [as require] (C:\Users\Maxi\Desktop\repo\esp8266_milight_hub\web\node_modules\natives\index.js:55:10)
    at Object.<anonymous> (C:\Users\Maxi\Desktop\repo\esp8266_milight_hub\web\node_modules\graceful-fs\fs.js:1:37)
    at Module._compile (internal/modules/cjs/loader.js:1156:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1176:10)
    at Module.load (internal/modules/cjs/loader.js:1000:32)
    at Function.Module._load (internal/modules/cjs/loader.js:899:14)
    at Module.require (internal/modules/cjs/loader.js:1042:19)
    at require (internal/modules/cjs/helpers.js:77:18)
b''
Encountered error CalledProcessError building webpage: Command '['node_modules\\.bin\\gulp.cmd']' returned non-zero exit status 1.
WARNING: Failed to build web package. Using pre-built page.
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (1.8.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 2.20402.4 (2.4.2)
 - tool-esptool 1.413.0 (4.13)
 - toolchain-xtensa 1.40802.0 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
ValueError: Invalid simple block '07010f2':
  File "C:\Users\Maxi\appdata\local\programs\python\python37\lib\site-packages\platformio\builder\main.py", line 169:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Maxi\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Maxi\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Maxi\.platformio\platforms\espressif8266\builder\main.py", line 316:
    target_elf = env.BuildProgram()
  File "C:\Users\Maxi\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Maxi\appdata\local\programs\python\python37\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Maxi\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Maxi\appdata\local\programs\python\python37\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Maxi\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Maxi\appdata\local\programs\python\python37\lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\Users\Maxi\appdata\local\programs\python\python37\lib\site-packages\platformio\builder\tools\piolib.py", line 886:
    if lm.get_package(spec):
  File "c:\users\maxi\appdata\local\programs\python\python37\lib\site-packages\platformio\package\manager\base.py", line 232:
    spec = self.ensure_spec(spec)
  File "c:\users\maxi\appdata\local\programs\python\python37\lib\site-packages\platformio\package\manager\base.py", line 103:
    return spec if isinstance(spec, PackageSpec) else PackageSpec(spec)
  File "c:\users\maxi\appdata\local\programs\python\python37\lib\site-packages\platformio\package\meta.py", line 119:
    self._parse(raw)
  File "c:\users\maxi\appdata\local\programs\python\python37\lib\site-packages\platformio\package\meta.py", line 222:
    raw = parser(raw)
  File "c:\users\maxi\appdata\local\programs\python\python37\lib\site-packages\platformio\package\meta.py", line 245:
    self.requirements = tokens[1].strip()
  File "c:\users\maxi\appdata\local\programs\python\python37\lib\site-packages\platformio\package\meta.py", line 162:
    else semantic_version.SimpleSpec(str(value))
  File "c:\users\maxi\appdata\local\programs\python\python37\lib\site-packages\semantic_version\base.py", line 618:
    self.clause = self._parse_to_clause(expression)
  File "c:\users\maxi\appdata\local\programs\python\python37\lib\site-packages\semantic_version\base.py", line 1014:
    return cls.Parser.parse(expression)
  File "c:\users\maxi\appdata\local\programs\python\python37\lib\site-packages\semantic_version\base.py", line 1034:
    raise ValueError("Invalid simple block %r" % block)
============================================= [FAILED] Took 14.49 seconds =============================================

Environment    Status    Duration
-------------  --------  ------------
nodemcuv2      FAILED    00:00:14.488
```

### Expected behavior

The webpage and firmware can be built.

### Setup information

* `pio --version`:  `PlatformIO, version 5.0.1`
* `npm --version`: 6.14.4
* `python --version`: 3.7.6

#### Firmware version

Master commit 27a1a6ac747eacc867cea8286fc691ce7e699e55
#### Output of http://milight-hub.local/about

Not applicable since uncompilable and unrunnable

#### Output of http://milight-hub.local/settings

Not applicable since uncompilable and unrunnable

### Additional context

I was made of this project and issue due to a user in our PlatformIO community forum reporting this build error (https://community.platformio.org/t/valueerror-invalid-simple-block-9d5f56fba1/15559/5?u=maxgerhardt). 

Your firmware is uncompilable (in the latest PIO Core 5) because of these two library definitions:

https://github.com/sidoh/esp8266_milight_hub/blob/27a1a6ac747eacc867cea8286fc691ce7e699e55/platformio.ini#L17-L22

Regarding `sidoh/WiFiManager#cmidgley` and `ratkins/RGBConverter@07010f2`.

Since PIO core 5, a new library repository system with a library owner syntax was established (`owner/libraryname`).  See [documentation](https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps). Since neither the `WiFiManager` nor the `RGBConverter` libraries are registered by their owners in the PIO library registry, installing them fails. 

Changing the syntax to just pull them from the github repo by using 

```
lib_deps_external = 
	WiFiManager=https://github.com/sidoh/WiFiManager.git#cmidgley
	RF24@~1.3.2
	ArduinoJson@~6.10.1
	PubSubClient@~2.7
	https://github.com/ratkins/RGBConverter.git
	WebSockets@~2.2.0
	CircularBuffer@~1.2.0
	PathVariableHandlers@~2.0.0
	RichHttpServer@~2.0.2
```
results in the firmware being able to be compiled. 

Website building is still broken; Since I'm not a webdev, no idea why.

maxgerhardt, 2020-10-03:
I think my website build issue is described here: https://stackoverflow.com/a/61405958/5296568

sidoh, 2020-10-09:
Thanks for digging into this, @maxgerhardt! Will try to take a look when I have more time.

sidoh, 2020-10-15:
Should be fixed by #659, thanks to @rjsachse!

closed 2020-10-15 by sidoh

lal12, 2020-11-07:
Hmm I guess the website trouble building part is not fixed. Is there any reason not to upgrade to gulp 4 as is suggested in the stackoverflow entry by maxgerhardt? I had to change the gulpfile a bit, but then it seems to work.

maxgerhardt, 2020-11-07:
You actually managed to build the website? What exact files did you use / replace?

lal12, 2020-11-07:
Well I updated to gulp 4 and adjusted the gulpfile.js a bit and it runs through sucessfully. Though I cannot test whether it really works, since the firmware build does not work for me, it throws a bunch of errors.

maxgerhardt, 2020-11-07:
Can you open another issue for that with the exact errors being thrown, and your PlatformIO core version (shown in the PIO home start screen in VSCode)? You can also open another issue for the website building and the fixed gulpfile

lal12, 2020-11-08:
Found a solution by now, adding `lib_compat_mode=strict` as suggested in the [forum post](https://community.platformio.org/t/valueerror-invalid-simple-block-9d5f56fba1/15559/15) which was posted above.