sidoh/esp8266_milight_hub #658: Ram on ESP8266EX 50kb  ( esp_wroom_02 , esp wroom , chinese clone esp )

mbd-shift, 2020-10-08:
### Describe the bug

incompatible with espressif8266EX

### Steps to reproduce

Upload, set up wifi, not get any further than that, no webclient.

### Expected behavior

Normal behaviour

### Setup information

#### Firmware version
1.10.6 all
1.5.0 all

**I think it fails due to the limit of only 50kb ram on this mcu.**
If you know where to change behaviour of the program considering ram usage, please let me know ...

sidoh, 2020-10-09:
I don't think I'm going to be able to cut the 25-30 KiB of RAM necessary to have it work with this chip. :(

I'd suggest grabbing another chip with the full 80 KiB -- fortunately they're super cheap!

closed 2020-10-09 by sidoh

mbd-shift, 2020-10-11:
> I'd suggest grabbing another chip with the full 80 KiB -- fortunately they're super cheap!

I don't know exactly where to find them... 
The market is so flooded with 12-F and other clones using esp8266ex chips.

Just got another fake in the mail even though the Article Said it was LOLIN ...

Can you maybe compile it for ESP8266EX? there must be a difference, since it's not working on 2 different boards, both with this chip

garmck, 2020-10-12:
> maybe a tiny version? with limitations on everything, no wifi manager etc?
> It's super hard to find the original module

To be honest I'm having a harder time finding your module.

I've bought numerous boards from here, these will suit what you require
https://www.aliexpress.com/item/32651747570.html
https://www.aliexpress.com/item/32665100123.html

mbd-shift, 2020-10-12:
> To be honest I'm having a harder time finding your module.
>
> I've bought numerous boards from here, these will suit what you require
> https://www.aliexpress.com/item/32651747570.html [...]

I have a bunch of modules that look exactly like the D1 you linked.
Was not worried at first, but after this code did not work, I decided to remove the shielding.
This revealed nothing more than what esptool would also tell me, but the MCU was the "ex" version with 50kb.

I'll order one from aliexpresss, see you in a month or so

sidoh, 2020-10-12:
I've never run into them, but last batch I ordered was probably close to a year ago. Sorry to hear you're being sent fakes. That's really crappy. :(

maxgerhardt, 2020-10-13:
>I think it fails due to the limit of only 50kb ram on this mcu.

I don't think that's the case. My NodeMCUv2 has a ESP12-F module which in turn has the **ESP8266EX** chip ([source](https://www.elecrow.com/download/ESP-12F.pdf)). The 50kB mentioned in [the datasheet](https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf) refers to what's left for the user when their SDK is used. This project only takes up 

```
RAM:   [=====     ]  50.2% (used 41136 bytes from 81920 bytes)
Flash: [=====     ]  47.6% (used 497568 bytes from 1044464 bytes)
```
of the ESP8266. Every ESP8266 module should have the same **RAM**. 

With [my fixes applied](https://github.com/sidoh/esp8266_milight_hub/issues/656) I uploaded the firmware to my NodeMCUv2 . It then auto-connected to my network (previously this ESP was already connected once), and on its IP address the HTTP server was available.

[image]

With the serial output being 

```
*WM: staticIP
*WM: Adding parameter
*WM: netmask
*WM: Adding parameter
*WM: gateway
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.172
Wifi connected succesfully
```

So I highly suggest that you look at the serial output of the module at 9600 baud and reset the module, to see what's being outputted. Because it *does* work on an ESP8266EX.