sidoh/esp8266_milight_hub #664: Why use NRF24L01+ and a ESP8266 and not a ESP8266-01s

VirtualSquash, 2020-10-31:
Hi everyone.

In advance I have not set up the hub yet because the parts for it are on the way.

But my Question is: why use the NRF24L01+ and not directly the NODMCU with his antenna for the hub?

And the second Question related to that is, why not use an ESP8266-01s all together?

Thanks for the Answers.

sidoh, 2020-10-31:
Although the esp8266 does use a 2.4 ghz radio, the apis are not low enough level to send the appropriate milight packets. I think the lowest level apis allow you to send 802.11 packets. (This is last I checked, which was about 2 yrs ago. Possible things have changed)

Since the nrf24 is required, so are SPI pins. The esp8266-01 does not break out SPI pins, so is not a good option. 

Hope that helps.

closed 2020-10-31 by sidoh

VirtualSquash, 2020-11-09:
Sorry for the long response time.
The Answer was exactly what i was looking for.

Thanks a lot.