sidoh/esp8266_milight_hub #665: Level and mode cannot be set via MQTT on FUT027

rfkd, 2020-11-02:
### Describe the bug

First of all, controlling my FUT027 (remote type RGB) via the MiLight Hub web interface completely works without any problems, thanks for that. Unfortunately I'm neither able to control the brightness of the LED strip nor receive (or change) the mode via MQTT, although controlling hue and state works just fine via MQTT.

When sending a specific level the LED brightness is actually increased by that level instead of it being applied directly. As sending negative levels has no impact I see no way to reduce the brightness via MQTT.

The "mode" key is not even part of the JSON string although it's enabled in the settings.

### Steps to reproduce

1. Select a color and change the brightness to 0 either via web interface or remote.
1. Set the level to 10 twice via MQTT (`mosquitto_pub -t milight/commands/0xABCD/rgb/1 -m '{"level":10}'`).
1. Change the mode via web interface or remote.

MQTT result is:
```
# Level set to 0 via remote
milight/states/0xABCD/rgb/0 {"state":"ON","status":"ON","brightness":0,"level":0,"hue":4,"color":"#FF1000","bulb_mode":"color"}

# Level set to 10 via MQTT: mosquitto_pub -t milight/commands/0xABCD/rgb/1 -m '{"level":10}'
milight/commands/0xABCD/rgb/1 {"level":10}
milight/updates/0xABCD/rgb/0 {"command":"brightness_up"}
milight/states/0xABCD/rgb/0 {"state":"ON","status":"ON","brightness":26,"level":10,"hue":4,"color":"#FF1000","bulb_mode":"color"}

# Level set to 10 via MQTT: mosquitto_pub -t milight/commands/0xABCD/rgb/1 -m '{"level":10}'
milight/commands/0xABCD/rgb/1 {"level":10}
milight/updates/0xABCD/rgb/0 {"command":"brightness_up"}
milight/states/0xABCD/rgb/0 {"state":"ON","status":"ON","brightness":51,"level":20,"hue":4,"color":"#FF1000","bulb_mode":"color"}

# Mode changed via remote
milight/updates/0xABCD/rgb/0 {"command":"next_mode"}
```

### Expected behavior

- Level stays at 10 (and could even be reduced by setting the level to 0 for example).
- The "mode" key is part of the MQTT message and gets updated after changing the mode on the remote.

### Setup information

#### Firmware version
See below.

#### Output of http://milight-hub.local/about

```json
{
	"firmware": "milight-hub",
	"version": "1.10.6",
	"ip_address": "x.x.x.x",
	"reset_reason": "External System",
	"variant": "nodemcuv2",
	"free_heap": 17248,
	"arduino_version": "2_4_2",
	"queue_stats": {
		"length": 0,
		"dropped_packets": 0
	}
}
```

#### Output of http://milight-hub.local/settings

```json
{
	"admin_username": "",
	"admin_password": "",
	"ce_pin": 4,
	"csn_pin": 15,
	"reset_pin": 0,
	"led_pin": -2,
	"radio_interface_type": "nRF24",
	"packet_repeats": 50,
	"http_repeat_factor": 1,
	"auto_restart_period": 0,
	"mqtt_server": "x.x.x.x",
	"mqtt_username": "",
	"mqtt_password": "",
	"mqtt_topic_pattern": "milight/commands/:device_id/:device_type/:group_id",
	"mqtt_update_topic_pattern": "milight/updates/:device_id/:device_type/:group_id",
	"mqtt_state_topic_pattern": "milight/states/:device_id/:device_type/:group_id",
	"mqtt_client_status_topic": "",
	"simple_mqtt_client_status": false,
	"discovery_port": 48899,
	"listen_repeats": 3,
	"state_flush_interval": 10000,
	"mqtt_state_rate_limit": 500,
	"mqtt_debounce_delay": 500,
	"packet_repeat_throttle_sensitivity": 0,
	"packet_repeat_throttle_threshold": 200,
	"packet_repeat_minimum": 3,
	"enable_automatic_mode_switching": false,
	"led_mode_wifi_config": "Fast blip",
	"led_mode_wifi_failed": "Fast toggle",
	"led_mode_operating": "Off",
	"led_mode_packet": "Off",
	"led_mode_packet_count": 3,
	"hostname": "milight-hub",
	"rf24_power_level": "MAX",
	"rf24_listen_channel": "LOW",
	"wifi_static_ip": "",
	"wifi_static_ip_gateway": "",
	"wifi_static_ip_netmask": "",
	"packet_repeats_per_loop": 10,
	"home_assistant_discovery_prefix": "",
	"wifi_mode": "n",
	"default_transition_period": 500,
	"rf24_channels": [
		"LOW",
		"MID",
		"HIGH"
	],
	"device_ids": [
		43981
	],
	"gateway_configs": [],
	"group_state_fields": [
		"state",
		"status",
		"brightness",
		"level",
		"hue",
		"saturation",
		"color",
		"mode",
		"kelvin",
		"color_temp",
		"bulb_mode",
		"computed_color",
		"effect",
		"device_id",
		"group_id",
		"device_type",
		"oh_color",
		"hex_color"
	],
	"group_id_aliases": {}
}
```

### Additional context
None.

PaulSzymanski, 2020-11-02:
I was also not able to set the brightness via mqtt. Instead of `{"command":"brightness_up"}` try `{"command":"level_up"}`. That was working for me. See #666 for a fix.

rfkd, 2020-11-03:
Actually I was not setting the "command" key at all, the "command" key from the log above was set by MiLight Hub in the update topic. However when setting the "command" key to "level_up/down" via `mosquitto_pub -t milight/commands/0xABCD/rgb/1 -m '{"command":"level_down"}'` as you suggested it's at least possible to control the brightness, so thanks for that. But shouldn't it be possible to directly set the "level" key as well (by using `mosquitto_pub -t milight/commands/0xABCD/rgb/1 -m '{"level":10}'`)?

By the way, setting the mode also works similar with e.g. `mosquitto_pub -t milight/commands/0xABCD/rgb/1 -m '{"command":"next_mode"}'`, thanks again.

sidoh, 2020-11-03:
It should be possible to set the value directly, with a few caveats:

* the RGB protocol does not have a command to set brightness directly
* the hub attempts to remedy this by issuing the appropriate number of up / down commands
* in order for this to work, brightness must be at a known value
* in order to bootstrap this, hub ramps brightness all the way down to 0 and then up to the appropriate value

looking at the code, it's possible something is broken with this, will take a look.

sidoh, 2020-11-24:
Does look like this was a bug -- will fix it in the next patch release (should go out this morning).

closed 2020-11-25 by sidoh