sidoh/esp8266_milight_hub #670: Can't compile project on PlatformIO?

PaulAtMedac, 2020-11-04:
Hi, this might turn out to be a stupid question or not to be project related at all, but I've tried around for a while now and nothing seems to be working so we'll see about that.
I have a nodemcuv2 which I'm trying to upload the project onto using platformio. The board definitely works as uploading the blink program works and I've recently used it to try out platformio's remote uploading feature. Connected to it is the NRF board.
Within PlatformIO (on VSCode, Win 10 btw), whenever I try compiling the project, I get a number of errors which I can't really make any sense of, so maybe one of the people here can help me out with em. Earlier today, the compiler couldn't find my Python install which I then reinstalled using the Windows Store, which then silenced the warning so I assume it worked correctly. 
I also tried just flashing the binary file, but then it wasn't creating any WiFi Network so I went back to compiling it myself, as maybe then I could actually debug it. Now, I am still getting these errors though:

(my username does have a blank space in it, but I don't think this is an issue as I've not yet run into an issue with that)

```
> Executing task in folder esp8266_milight_hub-master: C:\Users\<USERNAME>\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing nodemcuv2 (platform: espressif8266@~1.8; framework: arduino; board: nodemcuv2)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Attempting to build webpage...
npm WARN esp8266-milight-hub-web@1.0.0 No description
npm WARN esp8266-milight-hub-web@1.0.0 No repository field.

b'audited 429 packages in 3.976s\nfound 9 vulnerabilities (3 low, 6 high)\n  run `npm audit fix` to fix them, or `npm audit` for details\n'
b"[\x1b[90m19:31:24\x1b[39m] Using gulpfile E:\\Bibliotheken\\Documents\\PlatformIO\\Projects\\esp8266_milight_hub-master\\web\\gulpfile.js\n[\x1b[90m19:31:24\x1b[39m] Starting 'clean'...\n[\x1b[90m19:31:24\x1b[39m] Finished 'clean' after 4.23 ms\n[\x1b[90m19:31:24\x1b[39m] Starting 'buildfs_inline'...\n[\x1b[90m19:31:24\x1b[39m] Finished 'buildfs_inline' after 644 ms\n[\x1b[90m19:31:24\x1b[39m] Starting 'buildfs_embeded'...\n[\x1b[90m19:31:24\x1b[39m] Finished 'buildfs_embeded' after 35 ms\n[\x1b[90m19:31:24\x1b[39m] Starting 'default'...\n[\x1b[90m19:31:24\x1b[39m] Finished 'default' after 41 \xce\xbcs\n"
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (1.8.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 2.20402.4 (2.4.2)
 - tool-espotapy 1.1.3
 - tool-esptool 1.413.0 (4.13)
 - tool-mkspiffs 1.200.0 (2.0)
 - toolchain-xtensa 1.40802.0 (4.8.2)
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File "E:\Bibliotheken\Documents\PlatformIO\Projects\esp8266_milight_hub-master\.get_version.py", line 31, in <module>
    sys.stdout.write("-DMILIGHT_HUB_VERSION=%s %s" % (version.decode('utf-8'), ' '.join(sys.argv[1:])))
AttributeError: 'str' object has no attribute 'decode'
OSError: 'python3 .get_version.py' exited 1:
  File "C:\Users\<USERNAME>\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\<USERNAME>\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\<USERNAME>\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\<USERNAME>\.platformio\platforms\espressif8266@1.8.0\builder\main.py", line 316:
    target_elf = env.BuildProgram()
  File "C:\Users\<USERNAME>\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\<USERNAME>\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\<USERNAME>\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\<USERNAME>\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 124:
    env.ProcessFlags(env.get("BUILD_FLAGS"))
  File "C:\Users\<USERNAME>\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\<USERNAME>\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 222:
    env.Append(**env.ParseFlagsExtended(flags))
  File "C:\Users\<USERNAME>\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\<USERNAME>\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 185:
    for key, value in env.ParseFlags(str(raw)).items():
  File "C:\Users\<USERNAME>\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 815:
    do_parse(arg)
  File "C:\Users\<USERNAME>\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 681:
    arg = self.backtick(arg[1:])
  File "C:\Users\<USERNAME>\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 599:
    raise OSError("'%s' exited %d" % (command, status))
=========================================================================================== [FAILED] Took 10.47 seconds =========================================================================================== 
```

marcwoge, 2020-11-05:
I have the same problem...

```
PS E:\Marc\Downloads\esp8266_milight_hub-1.10.6\esp8266_milight_hub-1.10.6> platformio run -e nodemcuv2 --target upload
Processing nodemcuv2 (platform: espressif8266@~1.8; framework: arduino; board: nodemcuv2)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing tool-espotapy @ ~1.1.0
Tool Manager: tool-espotapy @ 1.1.3 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
INFORMATION: Es konnten keine Dateien mit dem angegebenen
Muster gefunden werden.
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (1.8.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 2.20402.4 (2.4.2)
 - tool-espotapy 1.1.3
 - tool-esptool 1.413.0 (4.13)
 - tool-mkspiffs 1.200.0 (2.0)
 - toolchain-xtensa 1.40802.0 (4.8.2)
INFORMATION: Es konnten keine Dateien mit dem angegebenen
Muster gefunden werden.
Traceback (most recent call last):
  File "E:\Marc\Downloads\esp8266_milight_hub-1.10.6\esp8266_milight_hub-1.10.6\.get_version.py", line 31, in <module>
    sys.stdout.write("-DMILIGHT_HUB_VERSION=%s %s" % (version.decode('utf-8'), ' '.join(sys.argv[1:])))
AttributeError: 'str' object has no attribute 'decode'
OSError: 'python3 .get_version.py' exited 1:
  File "C:\users\marc-\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\marc-\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\marc-\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\marc-\.platformio\platforms\espressif8266@1.8.0\builder\main.py", line 316:
    target_elf = env.BuildProgram()
  File "C:\Users\marc-\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\users\marc-\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\marc-\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\users\marc-\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 124:
    env.ProcessFlags(env.get("BUILD_FLAGS"))
  File "C:\Users\marc-\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\users\marc-\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 222:
    env.Append(**env.ParseFlagsExtended(flags))
  File "C:\Users\marc-\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\users\marc-\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 185:
    for key, value in env.ParseFlags(str(raw)).items():
  File "C:\Users\marc-\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 815:
    do_parse(arg)
  File "C:\Users\marc-\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 681:
    arg = self.backtick(arg[1:])
  File "C:\Users\marc-\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 599:
    raise OSError("'%s' exited %d" % (command, status))
================================================================================================== [FAILED] Took 3.02 seconds ================================================================================================== 

Environment    Status    Duration
-------------  --------  ------------
nodemcuv2      FAILED    00:00:03.016
============================================================================================= 1 failed, 0 succeeded in 00:00:03.016 =============================================================================================
```

sidoh, 2020-11-05:
Looks like the fallback logic for `.get_version.py` has some residual errors from the python2 => python3 migration. The directory you're running the build from is not a git repository, which means this script is unable to figure out what version it's compiling.

Should be an easy fix, but as a workaround, try replacing this line:

https://github.com/sidoh/esp8266_milight_hub/blob/master/.get_version.py#L31

with:

```python3
sys.stdout.write("-DMILIGHT_HUB_VERSION=%s %s" % ('UNKNOWN', ' '.join(sys.argv[1:])))
```

Alternatively, try checking the repository out from git.

sidoh, 2020-11-05:
This should be an easy fix, but can't work on it at this moment.

marcwoge, 2020-11-06:
Tanks for your advice... actually it doesn't work... 

But it was helpful anyway, I solved it by installing Git and connecting it with VSC... 
Got a few errors to solve and a burned ESP but now it works ... 👍

sidoh, 2020-11-24:
will be fixed in 1.10.7.

closed 2020-11-25 by sidoh