sidoh/esp8266_milight_hub #679: MQTT "white_mode" and "effect" OFF problem

Surtur, 2020-11-28:
Hi. 
How to disable in using MQTT "white_mode".
If "white_mode" is ON, the color cannot be changed using MQTT.
Everything works fine in the WEB GIU and not with MQTT.

sidoh, 2020-11-28:
Can't reproduce on my end. What remote type are you using?

MQTT log:

```
milight_test/commands/0x118d/rgb_cct/1 {"state": "ON", "color": {"r": 255, "g": 50, "b": 0}}
milight_test/state/0x118D/rgb_cct/1 {"status":"ON","bulb_mode":"color","hue":13,"saturation":100}
```

Surtur, 2020-11-29:
First turn on the light bulb in "white_mode" or "night_mode" or other effects
`milight_test/commands/0x118d/rgb_cct/1 {"state": "ON", "effect":"white_mode"}`

Then try to run it differently
`milight_test/commands/0x118d/rgb_cct/1 {"state": "ON", "color": {"r": 255, "g": 50, "b": 0}}`

Changing the mode is not working or something is missing from my code.

sidoh, 2020-11-29:
Yep, that's what I did. Maybe try posting your settings and full MQTT log?

poudenes, 2020-12-01:
ehmm, i see this error also sometimes. Whatever you try you don't get rid of the white led color. What i do is change bulb to a Kelvin color and then i can change it to a color. Otherwise its stick with the white led.
Im at my work, but when i have some time i will do some tests and try to repeat the problem

Surtur, 2020-12-01:
Everything is fine now.
I disabled the function in [Setings] -> [Radio] -> [Switch to previous mode after saturation/color commands] = <Disable>
Thank you for your time

closed 2020-12-01 by Surtur

sidoh, 2020-12-01:
Ah... gotcha. That setting has caused a lot of confusion. It is disabled by default, for what it's worth.