sidoh/esp8266_milight_hub #681: RF module does not respond after a couple of hours of work

domeesposito, 2020-11-30:
Hi @sidoh and all, first of all thank you for this very interesting and useful project! 

I successfully completed my set-up and all work fine. I've just one issue regarding the stability. I've already checked 
the troubleshooting guide but without having any suggestions. 

Issue: after a variable time, radio component does not respond anymore, also if i restart the nodemcu from UI menu. 
I also tried to sniff traditional milight radio but without success. It seems that after 2 or 3 hours, the radio component blocks all in/out signals. **The only way to solve it is to turn the device off and on again by cutting off the power cable.** 

What is your suggestion to solve it? 

Thank you, Domenico

- Board used: NodeMCU v3 and NRF24L01+ module

- Firmware: pre-compiled binary

-/about:

`{"version":"UNKNOWN","variant":"nodemcuv2","free_heap":28528,"arduino_version":"2_4_2","reset_reason":"Software/System restart"}`

- /settings: 

`{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":5,"http_repeat_factor":1,"auto_restart_period":60,"mqtt_server":"192.168.1.115:1883","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"milight/commands/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"","discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"device_ids":[1,2527,41120],"gateway_configs":[[27290,5987,5],[5378,5988,5],[2527,5989,5],[1,5990,5]],"group_state_fields":["state","level","hue","saturation","mode","color_temp","bulb_mode"]}`

camcs1, 2020-12-01:
I've had the same issue, I managed to improve longevity somewhat (~24hrs of uptime) by using a [shielded NRF24L01+ module ](https://www.aliexpress.com/item/4000132850562.html?spm=a2g0s.9042311.0.0.27424c4dD8aJUn). 

A few ideas I've had that I haven't got round to doing are 

- Separate 5v supply for the radio 

- 2 Hubs, one for receiving and the other for sending

I'll update you if either change improves things

domeesposito, 2020-12-02:
Thank you @camcs1 for the tips! The first one seems to be very simple and quick to test, I could try 👍 
I'll get back to you with the results.

camcs1, 2020-12-04:
@domeesposito Just a heads up the radio actually expects 2.0-3.6v

camcs1, 2020-12-14:
I've added one of these [breakout boards](https://www.aliexpress.com/item/32887624321.html?spm=a2g0s.9042311.0.0.27034c4dscfYPS) and combined with the shielded radio I've now not needed to reboot for a week. I've also taped the dupont connectors to stop them from working loose over time

sidoh, 2020-12-14:
Sounds like a loose SPI connection issue. This has come up quite a few times in the past (see https://github.com/sidoh/esp8266_milight_hub/issues/684 for a recent example).

Switch to soldered connections if you can. :)

domeesposito, 2021-05-21:
It works fine soldering connections! Thank you guys @sidoh @camcs1

closed 2021-05-21 by domeesposito