sidoh/esp8266_milight_hub #682: Compatibility with 4-zone Touch Panels

Teleportist, 2020-11-30:
Would someone kindly confirm if this project can be used to simply use the below wall panels as input controllers for my Home Assistant please?

Mains powered
https://www.aliexpress.com/item/32801790563.html

Battery powered
https://www.aliexpress.com/item/32816297066.html?mp=1

Thanks

sidoh, 2020-11-30:
Both of those should be supported, but can't confirm because I don't own them.

Should be able to use the MQTT features to use them as input. A related writeup can be found [here]. Not exactly what you're asking for, but should be able to adapt it easily enough.

Tontze, 2020-11-30:
I have those mains powered, they work, partly. I got only "ALL" button working, not 1, 2 ,3 or 4 button :/
To be noted, im not a coder and my abilities are limited, maybe someone else have them working.

garmck, 2020-11-30:
I have multiple B4 in my home.
As sidoh mentioned just use MQTT so sync the state - you will need to sniff the DeviceID of the controller and use that in your Home Assistant config.

Teleportist, 2020-12-01:
Thanks all for the replies.

Sidoh - There was no link in your post.

garmck - Thanks for confirming you have these working. I just need to build a NodeMCU based gateway and I'll get sniffing.

My plan is to use these panels to control a number of the below RGBW lamps, with the option to also control via the HA on mobile devices. So I need the RGB colour wheel and warm to cool white light slider working, as well as the dimmer slider and on/off channel and all buttons.

sidoh, 2020-12-01:
Sorry. This was supposed to be the link.

https://github.com/sidoh/esp8266_milight_hub/wiki/Using-Milight-Remote-with-HomeAssistant

Teleportist, 2021-01-26:
Just checking back in to confirm the hub is working flawlessly with my T4 MiLight 4-zone wall panel.

Setup was so simple, just connected the Node MCU to the NRF24L01+ and flashed with NodeMCU flasher.
(I did try PlatformIO, but found it too complicated)

I needed to do a little bit of tweaking to the MQTT payload in Node Red to get things just right, but now I have a very flexible wall panel capable of controlling my Tasmotised lighting dimmers, or even anything else connected to my Hassio.

Many thanks to you Sidoh for making this possible.

closed 2021-01-26 by Teleportist