sidoh/esp8266_milight_hub #684: Bulbs only respond after changing settings for nRF24 module

MichielvanBeers, 2020-12-07:
### Describe the bug

First of all: big thanks to all the work for this project! Very happy to be using this already for quite some time :) .

My bulbs already had the issue that they would turn off randomly, without any clear trigger, This was a bit inconvenient, but workable. Recently, my bulbs have also been stopping with responding to the MQTT messages send through HomeAssistant, changing the nRF24 Power Level, Listen Channel or Send Channel, makes them work for a brief time, but then they stop again. 

If someone could give me a pointer on a possible fix, I would be very happy!

### Steps to reproduce

1. Go to HomeAssistant, toggle lights (e.g. From off to on)
2. Notice that the a MQTT message is sent, but no response is received. State in HomeAssistant changes to ON though.
3. Notice that the lights don't turn on
4. Go to the Milight Hub web interface
5. Change nRF24 Power Level, Listen Channel or Send Channel to another setting
6. Click Submit
7. Toggle light in HomeAssistant
8. Notice that light does respond to toggle in HomeAssistant 
9. Wait 1-3 minutes
10. Repeat toggling
11. Notice that lights don't respond anymore

### Expected behavior

Lights toggle when MQTT messages is sent through HomeAsisstant

### Setup information

#### Firmware version
Issue first started on: 1.10.5
Current: 1.10.7

#### Output of http://milight-hub.local/about

```
{"firmware":"milight-hub","version":"1.10.7","ip_address":"192.168.1.147","reset_reason":"External System","variant":"nodemcuv2","free_heap":16536,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```
{"admin_username":"USER","admin_password":"PASS","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.1.15:1883","mqtt_username":"MQTT","mqtt_password":"Pietjepuk90-MQTT","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:hex_device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"mqtt_retain":true,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Off","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MIN","rf24_listen_channel":"MID","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"g","default_transition_period":1000,"rf24_channels":["MID"],"device_ids":[0,1,2,3,4,5,6],"gateway_configs":[[0,1001,6],[1,1002,6]],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{"Lamp bij de bank":["rgb_cct",0,1],"Lamp bij de kast":["rgb_cct",0,2],"Lamp boven het aanrecht":["rgb_cct",5,1],"Ledstrip onder bed":["rgb_cct",6,1]}}```

### Additional context

Things I already tried:

- Power cycling the HUB
- Moving the HUB closer to the lights (current distance from bulb 1 = 3 meters, bulb 2 = 1 meter)
- Updating the device to the latest firmware
- Changing the wifi setting from N to G

sidoh, 2020-12-07:
How are the ESP8266 and nRF24 connected? Sounds like maybe the connection's coming loose and reinitializing SPI fixes it.

MichielvanBeers, 2020-12-07:
That was it! Thanks so much for the quick response. The nRF24 module is connected through dupont cables and one came slightly loose. I'll make note to solder these together soon :) thanks!

closed 2020-12-07 by MichielvanBeers