sidoh/esp8266_milight_hub #685: Automation for remote-only commands

clau-bucur, 2020-12-10:
Hi,

I've set it up in HA, and added the automation from https://github.com/sidoh/esp8266_milight_hub/wiki/Using-Milight-Remote-with-HomeAssistant and it's working as expected.

I have however an use-case which this does not work properly:
- a light group is controlled by an automation triggered by motion detection: when motion is detected in that room, turn on the lights for a few minutes, at very low brightness;
- if the user uses the remote, to change the lights brightness for example, the above automation should be turned off, until the user turns off the lights from the remote. At that point, turn on the automation and let it control the lights.

Now, the problem here is that I get updates on the "MQTT update topic" when both the ESPMH controls the lights and when the remote controls the lights.
Correct, as the above doc says it so: "_If you define an update topic pattern (mqtt_update_topic_pattern) milight/updates/:device_id/:device_type/:group_id, both state changes sent by ESPMH, and captured packets sent from other devices will cause state updates to be published to the appropriate topic._"

My question would be, how could I filter out the "remote-only" commands so I can act only when the remote is operated?

Thanks.

sidoh, 2020-12-23:
Most milight bulbs can be paired with multiple remotes. You could pair them with the remote directly and a virtual ID used by HomeAssistant. HASS could react to the virtual ID commands, disabling the automation. I'm not advanced enough with HASS automations to know exactly what it'd look like, but should be possible.

clau-bucur, 2020-12-24:
I have found an independent solution using automation and script which works for other types of lights too: https://community.home-assistant.io/t/variables-in-script-condition-template/255159/12

closed 2020-12-24 by clau-bucur