sidoh/esp8266_milight_hub #686: Abrupt transition from OFF to ON

HX2003, 2020-12-10:
I am using the RGB+CCT radio, all works well except when I first set to either color/white mode, then send OFF command, then turn on but with a different color/white mode. Below are the commands I send in order.

1. `{ "state": "On", "kelvin": 0, "brightness": 255, "bulb_mode" : "white" }`
2. `{ "state": "Off"}`
3. `{ "state": "On", "hue": 1000, "saturation": 100, "brightness": 255, "bulb_mode" : "color" }`

What I observe is that the light ramps up to the white color, then switches to color mode. I am doing anything wrong?

sidoh, 2020-12-10:
Try disabling the "Switch to previous mode after saturation/color commands" setting under Radio.

HX2003, 2020-12-10:
> Try disabling the "Switch to previous mode after saturation/color commands" setting under Radio.

It's already disabled though.

sidoh, 2020-12-10:
Ah, maybe I misunderstood.

There is no way to go directly into color mode in your scenario, if that's what you're asking. Bulbs have to be turned on before the react to any other commands, and they always take on the state they were in before last being turned off.

closed 2020-12-11 by HX2003