sidoh/esp8266_milight_hub #687: Easy-to-use implementation in Google home environment

RealBadTexaner, 2020-12-13:
Many many thanks for that really great work of implementing the milight to an ESP 8266.

anyhow I have to admit as a noob on that topic, that I completely got stuck after setting up the ESP8266.

I managed to set it up, such that I can control my LED lights with the web gui. Moreover I like to implement all that now in control for Google home. Is there any easy to understand or easy-to-use implementation, which does not require understanding the technical background of MQTT or similar? Maybe an easy-to-use manual to connect to an already existing Google API

Many thanks in advance
Alex

camcs1, 2020-12-14:
I'm using [Home Assistant](https://www.home-assistant.io/) and [Nabu Casa](https://www.nabucasa.com/) to connect to Google Home. Nabu Casa is a paid service but it massively simplifies connecting to Google Home.

sidoh, 2020-12-14:
Those both sound like great options, thanks @camcs1!

@RealBadTexaner, I'm going to close this issue because I don't think there's anything actionable for the project here, but I'm happy to continue the conversation in the ticket.

I'd 2nd HomeAssistant, but it's a fair bit of effort to set that up as well.

closed 2020-12-14 by sidoh