sidoh/esp8266_milight_hub #688: Mqtt discovery automatically creating a device on home assistant

joryvd, 2020-12-17:
Hello,

I'm not too sure if it's an issue with the hub software or with home assistant. 
I'm using the hub with some bulbs around my house. I made aliasses for each bulb in the hub software. I control them through mqtt with home assistant (2020.12.1). However when I wanted to update a scene and wanted to add a bulb I noticed the hub was viewed as a device and all the lights are entities of this device. At first glance this wouldn't be too much of a problem. It did present an issue where it's not clear which bulbs are programmed in a scene since home assistant grabs all the entities from a device. I never noticed this problem in the past though.
If you need any more info please let me know!

Jory

sidoh, 2020-12-23:
Hi @joryvd!

I'm not entirely sure what the ask is here -- can you clarify what you'd like to be different?

Erriez, 2020-12-29:
Hi @joryvd,

I understand what you mean, because I had the same question:

When MiLight Hub is connected to Homeassistant with MQTT auto discovery enabled, the following Scene rule is applicable "If you add an entity that belongs to a device, the device will be added.". In other words, all registered lights in the MiLight hub (device) are added to the scene. 

As workaround, I configured all lights individually in the Homeassistant `configuration.yaml`. Then you can add individual lights to a scene. For example

**configuration.yaml**:
```yaml
mqtt:
  broker: pi
  port: 1883
  username: username
  password: password
  # discovery: true  <= Disabled
  #discovery_prefix: ha

light:
  # FUT092 0x1234
  - name: MiLight FUT092 Zone 0
    unique_id: milight_fut092_0
    command_topic: "milight/cmd/0x1234/rgb_cct/0"
    state_topic: "milight/state/0x1234/rgb_cct/0"
    
    # Use YAML anchor for common settings for other lights.
    <<: &MILIGHT_RGBCCT_PARAMS
      platform: mqtt
      schema: json
      brightness: true
      color_temp: true
      rgb: true
      effect: true
      effect_list: 
        - night_mode
        - white_mode
        - 0
        - 1
        - 2
        - 3
        - 4
        - 5
        - 6
        - 7
        - 8

  - name: MiLight FUT092 Zone 1
    unique_id: milight_fut092_1
    command_topic: "milight/cmd/0x1234/rgb_cct/1"
    state_topic: "milight/state/0x1234/rgb_cct/1"
    <<: *MILIGHT_RGBCCT_PARAMS
  - name: MiLight FUT092 Zone 2
    unique_id: milight_fut092_2
    command_topic: "milight/cmd/0x1234/rgb_cct/2"
    state_topic: "milight/state/0x1234/rgb_cct/2"
    <<: *MILIGHT_RGBCCT_PARAMS
  - name: MiLight FUT092 Zone 3
    unique_id: milight_fut092_3
    command_topic: "milight/cmd/0x1234/rgb_cct/3"
    state_topic: "milight/state/0x1234/rgb_cct/3"
    <<: *MILIGHT_RGBCCT_PARAMS
  - name: MiLight FUT092 Zone 4
    unique_id: milight_fut092_4
    command_topic: "milight/cmd/0x1234/rgb_cct/4"
    state_topic: "milight/state/0x1234/rgb_cct/4"
    <<: *MILIGHT_RGBCCT_PARAMS

  # FUT089 0xABCD
  - name: MiLight FUT089 Zone 0
    unique_id: milight_fut089_0
[... 119 lines cut ...]
    light.milight_fut091_zone_2:
      min_mireds: 153
      max_mireds: 500
      brightness: 107
      color_temp: 370
      friendly_name: MiLight FUT091 Zone 2
      supported_features: 43
      state: 'on'
    light.milight_fut091_zone_3:
      min_mireds: 153
      max_mireds: 500
      brightness: 133
      color_temp: 370
      friendly_name: MiLight FUT091 Zone 3
      supported_features: 43
      state: 'on'
    light.milight_fut091_zone_4:
      min_mireds: 153
      max_mireds: 500
      brightness: 61
      color_temp: 370
      friendly_name: MiLight FUT091 Zone 4
      supported_features: 43
      state: 'on'
```

joryvd, 2020-12-29:
Hi, 

Because of the holidays I forgot to reply sooner. Sorry for that. What @Erriez is describing is also what I'm experiencing. It did cross my mind to manually configure the lights instead of using the auto discovery function. That is indeed an option but one that requires more work. Especially when a substantial amount of lights will be added overtime it can get quite hefty. If that's the workaround then so be it, but if there's a solution that would be great.
I did notice when programming new scenes that although the milight "device" is getting added (so all the lights), only the lights I search for and click on when making the scene are actually stored. So it still enables me to make new scenes with only some of my lights, but requires me more to remember which ones I've clicked or work with YAML.

Erriez, 2020-12-30:
Would it be possible for MiLight Hub to register devices per address to Homeassistant with MQTT discovery enabled? In this case the amount of lights are already grouped per remote control and is easier to maintain.

closed 2023-12-01 by joryvd