sidoh/esp8266_milight_hub #691: Settings gone after reboot/unpower

DarkAllMan, 2020-12-21:
When I shutdown the board (restart/power down), the settings are gone.
I need to import the backup file and then it works. Is this software related, or is this a board issue?

### What firmware version(s) have you tried? 
1.10.7 (nodemcuv2)

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 
NodeMCUv3
### Output of http://milight-hub.local/about and http://milight-hub.local/settings 

#### /about
  |  
-- | --
firmware | "milight-hub"
version | "1.10.7"
ip_address | "192.168.1.3"
reset_reason | "Software/System restart"
variant | "nodemcuv2"
free_heap | 16040
arduino_version | "2_4_2"
queue_stats |  
length | 0
dropped_packets | 0

#### /settings
  |  
-- | --
admin_username | ""
admin_password | ""
ce_pin | 4
csn_pin | 15
reset_pin | 0
led_pin | -2
radio_interface_type | "nRF24"
packet_repeats | 50
http_repeat_factor | 1
auto_restart_period | 0
mqtt_server | ""
mqtt_username | ""
mqtt_password | ""
mqtt_topic_pattern | ""
mqtt_update_topic_pattern | ""
mqtt_state_topic_pattern | ""
mqtt_client_status_topic | ""
simple_mqtt_client_status | false
discovery_port | 48899
listen_repeats | 3
state_flush_interval | 10000
mqtt_state_rate_limit | 500
mqtt_debounce_delay | 500
mqtt_retain | true
packet_repeat_throttle_sensitivity | 0
packet_repeat_throttle_threshold | 200
packet_repeat_minimum | 3
enable_automatic_mode_switching | false
led_mode_wifi_config | "Fast toggle"
led_mode_wifi_failed | "On"
led_mode_operating | "Slow blip"
led_mode_packet | "Flicker"
led_mode_packet_count | 3
hostname | "Milight Hub"
rf24_power_level | "MAX"
rf24_listen_channel | "LOW"
wifi_static_ip | ""
wifi_static_ip_gateway | ""
wifi_static_ip_netmask | ""
packet_repeats_per_loop | 10
home_assistant_discovery_prefix | ""
wifi_mode | "n"
default_transition_period | 500
rf24_channels |  
0 | "LOW"
1 | "MID"
2 | "HIGH"
device_ids |  
0 | 1
1 | 2
2 | 3
3 | 4
4 | 5
5 | 6
6 | 7
gateway_configs |  
0 |  
0 | 1
1 | 1001
2 | 6
1 |  
0 | 2
1 | 1002
2 | 6
2 |  
0 | 3
1 | 1003
2 | 6
3 |  
0 | 4
1 | 1004
2 | 6
4 |  
0 | 5
1 | 1005
2 | 6
5 |  
0 | 6
1 | 1006
2 | 6
6 |  
0 | 7
1 | 1007
2 | 6
group_state_fields |  
0 | "state"
1 | "brightness"
2 | "mode"
3 | "color_temp"
4 | "bulb_mode"
5 | "computed_color"
group_id_aliases | {}

sidoh, 2020-12-23:
Hi @DarkAllMan ,

Does this happen consistently? If you change a setting and reboot the device, does it always clear settings?

DarkAllMan, 2020-12-27:
Today I retried again to simulate this behaviour and could not. Seems all is working ok now....

closed 2020-12-27 by DarkAllMan