sidoh/esp8266_milight_hub #694: Device ID for UDP settings

bcaro, 2020-12-26:
Hi!
I'am the author of a plugin for Jeedom and I plan to replace the official discontinued ibox1/2 by this bub.
In order to give advices to the users I need to know the meaning and the use of "Device ID" field in the configuration of the hub, because I do not use any ID in my plugin.
Thank's for your response

sidoh, 2020-12-26:
Hi @bcaro,

Device IDs are random 16 bit integers that are baked into every remote/wifi gateway. When you pair a bulb with a remote, it becomes associated with the remote's ID.

espMH is able to plug in whatever value you want for the device ID, which is why it's capable of controlling way more lights than the off the shelf remotes.

sidoh, 2020-12-26:
In the context of your plugin:

* New users can pick whatever ID they want
* Existing users can sniff traffic from their existing hub or remote and re-use the same ID to avoid the need to re-pair.

bcaro, 2020-12-28:
Thank you for the response
so for "enter hub ID" I put an hex like this : 0xAABB or an integer like this 12345
And in my plugin I use this ID in position 5 and 6 of the frame :
0x80, 0x00, 0x00, 0x00, 0x11, 0xFF, 0xFF, 0xE6, 0x80, 0x00, 0x31
Thank's 
Bernard

sidoh, 2020-12-28:
Yeah, that sounds right.

For what it's worth -- if it's possible for your plugin to use something other than the UDP protocol, I'd highly recommend it. MQTT is best, but requires your users to have an MQTT broker set up.

bcaro, 2020-12-28:
OK thanks
I will keep the udp protocol for users who have configured their devices et propose REST API for new users. Thank's.

closed 2020-12-28 by sidoh