sidoh/esp8266_milight_hub #695: Immediate state return when using a remote ?

bcaro, 2020-12-28:
Hi!
It seems that, using MQTT, when the state of a lamp is changed using a remote, a message is sent immediately.
Is it the same case with the rest API ? or using mdns ? (I think it is not the case using UDP protocol as Milight does not).
Thank's

sidoh, 2020-12-28:
Not totally sure I understand the question, so let me know if I'm not getting it --

* When configured, state updates are sent via MQTT regardless of how the command was received. REST, UDP, and MQTT commands will all trigger an MQTT state update
* The UDP protocol is strictly 1-way. No way to get state updates with it
* The REST protocol _can_ return the state of a light when sending a command, but doesn't by default (this is done to make requests asynchronous)

bcaro, 2020-12-28:
Ok thanks ! mqtt for 2 ways protocol

sidoh, 2020-12-28:
👍

closed 2020-12-28 by sidoh