sidoh/esp8266_milight_hub #698: How to change WiFi SSID / password

Erriez, 2020-12-31:
Thanks for creating this great project!

What is the easiest way to change WiFi SSID / password?

sidoh, 2020-12-31:
Hi @Erriez , are you referring to the setup password? That's hardcoded unfortunately.

Erriez, 2020-12-31:
Hi @sidoh, Thanks for your fast reply. 

> are you referring to the setup password? That's hardcoded unfortunately.

No, I want to switch to another WiFi access point, because I encounter connection stability issues between Milighthub and MQTT/Homeassistant.

No problem for hard-coded access point password.

In the mean time I found the page `Admin | Actions | Clear Wifi Config`. When I hit this button, the serial console prints:

```
Resetting Wifi and then Restarting...
```

Nothing happens. I have to press the reset button manually. Then the access point is available to enter a different SSID / password.

closed 2020-12-31 by Erriez