sidoh/esp8266_milight_hub #699: Homeassistant scene RGB or white not saved

Erriez, 2020-12-31:
I've a setup with RGB+CCT LED'strips, spots and bulb which can be controlled via the MiLight webinterface. Now I'm creating scenes in Homeassistant. Switching between scenes does not switch between RGB and white state. Is this a configuration setting?

**milight-hub/settings**
```yaml
admin_username | ""
admin_password | ""
ce_pin | 0
csn_pin | 15
reset_pin | 0
led_pin | -2
radio_interface_type | "nRF24"
packet_repeats | 200
http_repeat_factor | 1
auto_restart_period | 0
mqtt_server | "192.168.0.249"
mqtt_username | "****"
mqtt_password | "****"
mqtt_topic_pattern | "milight/cmd/:device_type/:device_id/:group_id"
mqtt_update_topic_pattern | "milight/update/:device_type/:device_id/:group_id"
mqtt_state_topic_pattern | "milight/state/:device_type/:device_id/:group_id"
mqtt_client_status_topic | ""
simple_mqtt_client_status | false
discovery_port | 48899
listen_repeats | 1
state_flush_interval | 10000
mqtt_state_rate_limit | 200
mqtt_debounce_delay | 200
mqtt_retain | true
packet_repeat_throttle_sensitivity | 0
packet_repeat_throttle_threshold | 200
packet_repeat_minimum | 3
enable_automatic_mode_switching | false
led_mode_wifi_config | "Flicker"
led_mode_wifi_failed | "On"
led_mode_operating | "Off"
led_mode_packet | "Fast blip"
led_mode_packet_count | 2
hostname | "milight-hub"
rf24_power_level | "MAX"
rf24_listen_channel | "LOW"
wifi_static_ip | ""
wifi_static_ip_gateway | ""
wifi_static_ip_netmask | ""
packet_repeats_per_loop | 10
home_assistant_discovery_prefix | ""
wifi_mode | "n"
default_transition_period | 500
rf24_channels |  
0 | "LOW"
1 | "HIGH"
device_ids |  
0 | 6098
gateway_configs | []
group_state_fields |  
0 | "state"
1 | "brightness"
2 | "color"
3 | "mode"
4 | "color_temp"
5 | "effect"
group_id_aliases |  
```

**MQTT status**
* state
* brightness
* color
* mode
* color_temp
* effect

**Homeassistant configuration.yaml**
```yaml
light:
  #-----------------------------------------------------------------------------
  # Bedroom Erwin
  #-----------------------------------------------------------------------------
  - name: MiLight FUT092 0x17D2 Zone 1
    unique_id: milight_fut092_0x17D2_1
    command_topic: "milight/cmd/rgb_cct/0x17D2/1"
    state_topic: "milight/state/rgb_cct/0x17D2/1"
    
    # Use YAML anchor for common settings for other lights.
    <<: &MILIGHT_RGBCCT_PARAMS
      platform: mqtt
      schema: json
      brightness: true
      color_temp: true
      rgb: true
      effect: true
      effect_list: 
        - night_mode
        - white_mode
```

**Homeassistant scenes.yaml**
```yaml
- id: '1609275355223'
  name: Scene white
  entities:
    light.milight_fut092_0x17D2_zone_1:
      min_mireds: 153
      max_mireds: 500
      effect_list:
      - night_mode
      - white_mode
      brightness: 117
      color_temp: 333
      hs_color:
      - 240
      - 100
      rgb_color:
      - 0
      - 0
      - 255
      xy_color:
      - 0.136
      - 0.04
      friendly_name: MiLight LED strip
      supported_features: 63
      state: 'on'
- id: '1609275576445'
  name: Scene RGB colorful
  entities:
    light.milight_fut092_0x17D2_zone_1:
      min_mireds: 153
      max_mireds: 500
      effect_list:
      - night_mode
      - white_mode
      brightness: 54
      color_temp: 216
      hs_color:
      - 240
      - 100
      rgb_color:
      - 0
      - 0
      - 255
      xy_color:
      - 0.136
      - 0.04
      friendly_name: MiLight LED strip
      supported_features: 63
      state: 'on'
```

How can a Homeassistant scene switch between RGB and white mode?

Erriez, 2021-01-01:
After more research, the problem is located in Homeassistant:

https://github.com/home-assistant/frontend/issues/8062

The root cause is that Homeassistant saves an `effect_list:` with all effects listed in the drop-down instead of one selected `effect:`.

I assume nobody is able to switch between RGB and white mode with Homeassistant. Please correct me if I'm wrong.

GodGMN, 2022-12-30:
Hello, apparently I'm having this error too. If I'm in RGB mode, it **will** switch to white temperature mode, however it won't switch back.

I have two automations: one sets the color to purple at 33% brightness (which is RGB mode), the other one sets the color to cold white at 100% brightness (which is temperature mode).

When I have let's say red lights, if I press the first automation **it will** switch to purple and 33 brightness. If I press the second, it will also change to bright white (proper white, the RGB mode white is blueish and much darker).

However, if the light bulb is currently set to temperature mode, it **will not** switch back to color mode through automations, I have to manually select color mode again and pick a color. If I use the first automation while my light bulb is using the temperature mode, it will only go 33% brightness, but color won't change at all.

This is my yaml automations file:

```
- id: '1672376921722'
  name: Gaming
  entities:
    light.lamparita:
      min_color_temp_kelvin: 2000
      max_color_temp_kelvin: 6535
      min_mireds: 153
      max_mireds: 500
      supported_color_modes:
      - brightness
      - color_temp
      - hs
      color_mode: hs
      brightness: 57
      hs_color:
      - 299.833
      - 100
      rgb_color:
      - 254
      - 0
      - 255
      xy_color:
      - 0.384
      - 0.154
      friendly_name: Lamparita
      supported_features: 0
      state: 'on'
  icon: mdi:laptop
  metadata: {}
- id: '1672377096598'
  name: Light
  entities:
    light.lamparita:
      min_color_temp_kelvin: 2000
      max_color_temp_kelvin: 6535
      min_mireds: 153
      max_mireds: 500
      supported_color_modes:
      - color_temp
      color_mode: color_temp
      brightness: 255
      xy_color:
      - 0.331
      - 0.335
      friendly_name: Lamparita
      supported_features: 0
      color_temp_kelvin: 6329
      color_temp: 158
      state: 'on'
  icon: mdi:white-balance-sunny
  metadata: {}
```