sidoh/esp8266_milight_hub #703: All command should set the state of the 4 lights

DerrickR8, 2021-01-08:
### Describe the bug

First off, AWESOME program.  Thanks a lot.

Using the ALL group (0) does not update the MQTT state of groups 1-4.

The code for GroupStateStore::set() in lib\MiLightState\GroupStateStore.cpp looks like it loops through the 4 groups if group 0 is set, but only group 0 updates.

From MQTT Explorer, only 0 updates it state.  Also the 1-4 states do not update on Home Assistant or Google Home.

`milight
states
0x2
cct
rgbw
1 = {"state":"ON","brightness":255,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
3 = {"state":"OFF","brightness":255,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
2 = {"state":"ON","brightness":255,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
4 = {"state":"ON","brightness":255,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
0 = {"state":"OFF","color":{"r":255,"g":255,"b":255}}`

### Steps to reproduce

Turn Group All (0) on or off.

### Expected behavior

State of Groups 1-4 should update in MQTT

### Setup information

#### Firmware version
nodemcuv2-1.10.7

#### Output of http://milight-hub.local/about

```json
{"firmware":"milight-hub","version":"1.10.7","ip_address":"192.168.0.222","reset_reason":"External System","variant":"nodemcuv2","free_heap":16144,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```json
{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":150,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.0.222:1883","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"mqtt_retain":true,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":150,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"Fast blip","led_mode_operating":"Off","led_mode_packet":"Off","led_mode_packet_count":3,"hostname":"hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"pre","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[1,4468,2,1174],"gateway_configs":[],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{"Bathroom":["cct",2,1],"Colour Group 1":["rgbw",2,0],"Den":["cct",2,2],"Dining Room":["rgbw",2,1],"Front Light":["rgbw",1,1],"Hallway":["rgbw",2,3],"Kitchen":["rgbw",2,2],"Living Room":["rgb_cct",1,1],"Master Bedroom":["rgbw",2,4],"White Group 1":["cct",2,0]}}
```

### Additional context

allmazz, 2021-01-12:
I confirm, I also have this problem

sidoh, 2021-01-12:
Hi @DerrickR8, this is intentional behavior. My reasoning is that updating up to 8 groups for each command sent to group 0 is likely to cause significant strain on the ESP8266. Internal state for the individual groups is updated, but not broadcasted via MQTT.

You could implement a fan-out listener in something like HomeAssistant or Node-RED that does this for you.

DerrickR8, 2021-01-15:
OK Thanks.  Node-Red solved the issue.
Not sure why it would not accept a wildcard for the remote type.  Says disconnected when I do.
milight/states/0x2/+/0

I'll see if I can get Node-Red to send my remote signals to lights, because I could not get the Home Assistant action to work as shown on the Wiki-page.  Off topic though.

I attached a pic to help any others,

[image]

ccutrer, 2021-04-01:
@sidoh : the problem with the current implementation is that if a listener _reconnects_ (i.e. OpenHAB restarts), it will not know how to process the group 0 state, because processing it correctly depends on if group 0 was applied last, or an actual group was applied last. Sending 4 or 8 extra MQTT messages is not very taxing - TCP over wifi is _far_ faster than the rate limiting required on the milight radio. Even if you're using QoS 1 and waiting for the PUBACK after each publish, 8 messages is trivially fast with a local MQTT server. Though ideally if you're using QoS 1 you'll batch all the publishes before waiting for any of the PUBACKs. I have zero problems publishing dozens of messages effectively instantly to MQTT (over TLS!) with QoS 1 from an ESP8266 for another project (using the NodeMCU MQTT library).

sidoh, 2021-04-01:
@ccutrer, this is not something I'm going to get to myself. If you want to take a stab and do some stress-testing, I'd welcome a PR!

ccutrer, 2021-04-01:
Sounds good. It might take me a few weeks - I'm finishing a theater room, and the lights are only one of many sub-projects.