sidoh/esp8266_milight_hub #712: Null color being sent

golddragon007, 2021-02-22:
### Describe the bug & Steps to reproduce
In color bulb mode if we set to red the color and the saturation down to 0, null color will be sent.

### Expected behavior
Proper hex color.

### Setup information

#### Firmware version

1.10.7

#### Output of http://milight-hub.local/about

```json
{
"firmware": "milight-hub",
"version": "1.10.7",
"ip_address": "",
"reset_reason": "External System",
"variant": "nodemcuv2",
"free_heap": 16048,
"arduino_version": "2_4_2",
"queue_stats": {
"length": 0,
"dropped_packets": 0
}
}
```

#### Output of http://milight-hub.local/settings

```json
{
"admin_username": "",
"admin_password": "",
"ce_pin": 4,
"csn_pin": 15,
"reset_pin": 0,
"led_pin": -2,
"radio_interface_type": "nRF24",
"packet_repeats": 50,
"http_repeat_factor": 1,
"auto_restart_period": 0,
"mqtt_server": "",
"mqtt_username": "",
"mqtt_password": "",
"mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
"mqtt_update_topic_pattern": "milight/updates/:device_id/:device_type/:group_id",
"mqtt_state_topic_pattern": "milight/states/:device_id/:device_type/:group_id",
"mqtt_client_status_topic": "stat/milight",
"simple_mqtt_client_status": false,
"discovery_port": 48899,
"listen_repeats": 3,
"state_flush_interval": 10000,
"mqtt_state_rate_limit": 500,
"mqtt_debounce_delay": 500,
"mqtt_retain": false,
"packet_repeat_throttle_sensitivity": 0,
"packet_repeat_throttle_threshold": 200,
"packet_repeat_minimum": 3,
"enable_automatic_mode_switching": false,
"led_mode_wifi_config": "Fast toggle",
"led_mode_wifi_failed": "On",
"led_mode_operating": "Slow blip",
"led_mode_packet": "Flicker",
"led_mode_packet_count": 3,
"hostname": "milight-hub",
"rf24_power_level": "MAX",
"rf24_listen_channel": "LOW",
"wifi_static_ip": "",
"wifi_static_ip_gateway": "",
"wifi_static_ip_netmask": "",
"packet_repeats_per_loop": 10,
"home_assistant_discovery_prefix": "homeassistant",
"wifi_mode": "n",
"default_transition_period": 500,
"rf24_channels": [
"LOW",
"MID",
"HIGH"
],
"device_ids": [],
"gateway_configs": [],
"group_state_fields": [
"state",
"status",
"brightness",
"level",
"hue",
"saturation",
"color",
"mode",
"kelvin",
"color_temp",
"bulb_mode",
"computed_color",
"effect",
"device_id",
"group_id",
"device_type",
"oh_color",
"hex_color"
],
"group_id_aliases": {}
}
```

### Additional context
Happens at hue value 359-1

sidoh, 2021-02-25:
If possible, please paste a REST or MQTT command you're using to produce this issue.

golddragon007, 2021-02-26:
I touching my FUT089 remote and setting from there these values, it's the red color. And when I do that I get in MQTT status message with null color.

sidoh, 2021-02-26:
A full copy/paste of your MQTT log would be really helpful :)

sidoh, 2021-02-26:
(I use `mosquitto_sub` for this personally)

golddragon007, 2021-02-26:
I receive this from the ESP:
```
{
  "state": "ON",
  "status": "ON",
  "brightness": 255,
  "level": 100,
  "hue": 1,
  "saturation": 100,
  "color": null,
  "bulb_mode": "color"
}
```

If I select the red color on the hue wheel.

sidoh, 2021-03-04:
I don't know why it's only affecting `oh_color` and `hex_color`, but it is. Try disabling those two (they're all mutually exclusive with each other) and keeping `color`.

Definitely an issue, but don't have time to dig in right now.

Relevant code is here, if anyone wants to take a crack:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLightState/GroupState.cpp#L819

golddragon007, 2021-03-05:
How can I debug this or dump out data?

sidoh, 2021-03-06:
OK, I think this is just happening because the JSON buffer is running out of room. It's a static 200 bytes, and includes some overhead in addition to the raw fields. The generated state message is large enough that it's not able to allocate memory to set more fields.

Try de-selecting some of the group state fields. In particular, you probably don't need both `state` and `status`, or any of `device_id`, `device_type` or `group_id`.

closed 2021-03-06 by sidoh