sidoh/esp8266_milight_hub #719: FUT035 CCT led strip controller compatibility confirmed

sedlacekdavid, 2021-03-11:
Hi, I'd like to confirm that I can pair/control a FUT035 controller. What a great project!

sedlacekdavid, 2021-03-11:
[image]
[image]

spectroman, 2021-10-05:
@sedlacekdavid any chance for you to make available the 3d printer project and the schema for the antenna? it would be awesome to have the plans u did.

xbmcnut, 2023-09-10:
> Hi, I'd like to confirm that I can pair/control a FUT035 controller. What a great project!

Hi. Very old post but could you possibly indicate what settings you used to pair? (Group/Remote Type)